yanshouwang/bluetooth_low_energy #13: Invalid UUID string

spiderman16888, 2023-09-08:
v2.0.3 system show this error:

======== Exception caught by services library ======================================================
The following FormatException was thrown during a platform message callback:
Invalid UUID string

When the exception was thrown, this was the stack: 
#0      new UUID.fromString (package:bluetooth_low_energy_platform_interface/src/uuid.dart:66:7)
#1      _extension#0.toAdvertisement.<anonymous closure> (package:bluetooth_low_energy_darwin/src/my_central_controller.dart:330:26)
#2      MapBase.map (dart:collection/maps.dart:82:28)
#3      _extension#0.toAdvertisement (package:bluetooth_low_energy_darwin/src/my_central_controller.dart:328:68)
#4      MyCentralController.onDiscovered (package:bluetooth_low_energy_darwin/src/my_central_controller.dart:290:47)
#5      MyCentralControllerFlutterApi.setup.<anonymous closure> (package:bluetooth_low_energy_darwin/src/my_api.g.dart:686:15)
#6      BasicMessageChannel.setMessageHandler.<anonymous closure> (package:flutter/src/services/platform_channel.dart:212:49)
#7      _DefaultBinaryMessenger.setMessageHandler.<anonymous closure> (package:flutter/src/services/binding.dart:567:35)

yanshouwang, 2023-09-25:
Yes, there are some uuids not handled correct, I think I will fix this some days later.

yanshouwang, 2023-10-09:
The latest [3.0.0-dev.3](https://pub.dev/packages/bluetooth_low_energy/versions/3.0.0-dev.3) should resolve this.

spiderman16888, 2023-10-12:
fixed in 3.0.0. thanks

closed 2023-10-12 by spiderman16888