yanshouwang/bluetooth_low_energy #17: `IllegalStateException` when connecting.

flukejones, 2023-09-19:
I'm not sure what triggered this exactly, it may have been a connection attempt after a period of time.

The error log is:
```
E/flutter (11978): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(IllegalStateException, java.lang.IllegalStateException: Connect failed with status: 133, Cause: null, Stacktrace: java.lang.IllegalStateException: Connect failed with status: 133
E/flutter (11978): 	at dev.yanshouwang.bluetooth_low_energy_android.MyCentralController.onConnectionStateChange(MyCentralController.kt:497)
E/flutter (11978): 	at dev.yanshouwang.bluetooth_low_energy_android.MyBluetoothGattCallback.onConnectionStateChange$lambda$0(MyBluetoothGattCallback.kt:14)
E/flutter (11978): 	at dev.yanshouwang.bluetooth_low_energy_android.MyBluetoothGattCallback.$r8$lambda$iRmWtAhPoCVccasdkVRdGy0p5Yo(Unknown Source:0)
E/flutter (11978): 	at dev.yanshouwang.bluetooth_low_energy_android.MyBluetoothGattCallback$$ExternalSyntheticLambda3.run(Unknown Source:8)
E/flutter (11978): 	at android.os.Handler.handleCallback(Handler.java:942)
E/flutter (11978): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/flutter (11978): 	at android.os.Looper.loopOnce(Looper.java:226)
E/flutter (11978): 	at android.os.Looper.loop(Looper.java:313)
E/flutter (11978): 	at android.app.ActivityThread.main(ActivityThread.java:8757)
E/flutter (11978): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter (11978): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
E/flutter (11978): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
E/flutter (11978): , null)
E/flutter (11978): #0      MyCentralControllerHostApi.connect (package:bluetooth_low_energy_android/src/my_api.g.dart:351:7)
E/flutter (11978): <asynchronous suspension>
E/flutter (11978): #1      MyCentralController.connect (package:bluetooth_low_energy_android/src/my_central_controller.dart:107:5)
E/flutter (11978): <asynchronous suspension>
E/flutter (11978):
```

flukejones, 2023-09-19:
There are many areas where rethrowing or passing the error to dart would be helpful:

```
E/flutter (25967): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(IllegalStateException, java.lang.IllegalStateException: GATT is disconnected with status: 0, Cause: null, Stacktrace: java.lang.IllegalStateException: GATT is disconnected with status: 0
E/flutter (25967): 	at m3.s.v(Unknown Source:48)
E/flutter (25967): 	at m3.p.q(Unknown Source:12)
E/flutter (25967): 	at m3.p.j(Unknown Source:0)
E/flutter (25967): 	at m3.h.run(Unknown Source:8)
E/flutter (25967): 	at android.os.Handler.handleCallback(Handler.java:942)
E/flutter (25967): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/flutter (25967): 	at android.os.Looper.loopOnce(Looper.java:226)
E/flutter (25967): 	at android.os.Looper.loop(Looper.java:313)
E/flutter (25967): 	at android.app.ActivityThread.main(ActivityThread.java:8757)
E/flutter (25967): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter (25967): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
E/flutter (25967): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
E/flutter (25967): , null)
E/flutter (25967): #0      MyCentralControllerHostApi.discoverGATT (package:bluetooth_low_energy_android/src/my_api.g.dart:422)
E/flutter (25967): <asynchronous suspension>
E/flutter (25967): #1      MyCentralController.discoverGATT (package:bluetooth_low_energy_android/src/my_central_controller.dart:134)
E/flutter (25967): <asynchronous suspension>
```

```
E/flutter (25649): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(IllegalStateException, java.lang.IllegalStateException, Cause: null, Stacktrace: java.lang.IllegalStateException
E/flutter (25649): 	at m3.s.l(Unknown Source:86)
E/flutter (25649): 	at m3.q0$a.t(Unknown Source:45)
E/flutter (25649): 	at m3.q0$a.i(Unknown Source:0)
E/flutter (25649): 	at m3.o0.a(Unknown Source:2)
E/flutter (25649): 	at a4.a$b.a(Unknown Source:17)
E/flutter (25649): 	at o3.c.l(Unknown Source:18)
E/flutter (25649): 	at o3.c.m(Unknown Source:40)
E/flutter (25649): 	at o3.c.i(Unknown Source:0)
E/flutter (25649): 	at o3.b.run(Unknown Source:12)
E/flutter (25649): 	at android.os.Handler.handleCallback(Handler.java:942)
E/flutter (25649): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/flutter (25649): 	at android.os.Looper.loopOnce(Looper.java:226)
E/flutter (25649): 	at android.os.Looper.loop(Looper.java:313)
E/flutter (25649): 	at android.app.ActivityThread.main(ActivityThread.java:8757)
E/flutter (25649): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter (25649): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
E/flutter (25649): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
E/flutter (25649): , null)
E/flutter (25649): #0      MyCentralControllerHostApi.connect (package:bluetooth_low_energy_android/src/my_api.g.dart:351)
E/flutter (25649): <asynchronous suspension>
E/flutter (25649): #1      MyCentralController.connect (package:bluetooth_low_energy_android/src/my_central_controller.dart:107)
E/flutter (25649): <asynchronous suspension>
```

flukejones, 2023-09-20:
Because these errors are not caught and thrown in to dart space, I have to use a timer to check if I've successfully connected are characteristic or not and abort/restart connection. It's quite an inconvenient way to do things.

yanshouwang, 2023-09-25:
Can you provide a sample code here? I think you can try catch this exception in flutter side, it's a normal throw when connect to a peripheral, and it's a necessary to try catch when call connect and other apis.

flukejones, 2023-09-25:
My apologies. We were missing a try/catch in one final block of code.

closed 2023-09-25 by flukejones