yanshouwang/bluetooth_low_energy #22: android device: requestMtu issue

spiderman16888, 2023-10-20:
D/BluetoothGatt(24178): configureMTU() - device: D029BA_C mtu: 517
D/BluetoothGatt(24178): onConnectionUpdated() - Device=D029BA_C interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt(24178): onConfigureMTU() - Device=D029BA_C mtu=247 status=4
I/flutter (24178): error: PlatformException(IllegalStateException, java.lang.IllegalStateException: Request MTU failed with status: 4, Cause: null, Stacktrace: java.lang.IllegalStateException: Request MTU failed with status: 4
I/flutter (24178): 	at dev.yanshouwang.bluetooth_low_energy_android.MyCentralManager.onMtuChanged(MyCentralManager.kt:522)
I/flutter (24178): 	at dev.yanshouwang.bluetooth_low_energy_android.MyBluetoothGattCallback.onMtuChanged$lambda$1(MyBluetoothGattCallback.kt:21)
I/flutter (24178): 	at dev.yanshouwang.bluetooth_low_energy_android.MyBluetoothGattCallback.$r8$lambda$YuSPH4xgjMukXLzXKi-mkORqrJ4(Unknown Source:0)
I/flutter (24178): 	at dev.yanshouwang.bluetooth_low_energy_android.MyBluetoothGattCallback$$ExternalSyntheticLambda5.run(Unknown Source:8)
I/flutter (24178): 	at android.os.Handler.handleCallback(Handler.java:942)

yanshouwang, 2023-10-20:
This means the MTU request is failed, I called requestMtu after GATT discovered in the newest version, and if the request is failed, the discoverGATT will throw this error, will fix this behavior in the next version.
the status 4 seems releated with multiple MTU requests, look at this: [status = 4 error when calling requestMtu for twice in Android APP](https://devzone.nordicsemi.com/f/nordic-q-a/37163/status-4-error-when-calling-requestmtu-for-twice-in-android-app)
I suggest to use version 3.0.1 if you can't resolve this error, this will be fixed as soon as possible.

closed 2023-10-21 by yanshouwang

spiderman16888, 2023-11-10:
I use v4.0.0 version, then error: BLE can't access..

D/BluetoothGatt(18048): discoverServices() - device: D029BA_C
D/BluetoothGatt(18048): onSearchComplete() = Device=D029BA_C Status=0
E/IMGMapper(18048): set:488 set: Unset optional value from type SMPTE2086
E/IMGMapper(18048): set:488 set: Unset optional value from type CTA861_3
D/BluetoothGatt(18048): configureMTU() - device: D029BA_C mtu: 517
D/BluetoothGatt(18048): onConfigureMTU() - Device=D029BA_C mtu=247 status=4
E/IMGMapper(18048): set:488 set: Unset optional value from type SMPTE2086
E/IMGMapper(18048): set:488 set: Unset optional value from type CTA861_3
E/IMGMapper(18048): set:488 set: Unset optional value from type SMPTE2086
E/IMGMapper(18048): set:488 set: Unset optional value from type CTA861_3
E/IMGMapper(18048): set:488 set: Unset optional value from type SMPTE2086
E/IMGMapper(18048): set:488 set: Unset optional value from type CTA861_3
E/IMGMapper(18048): set:488 set: Unset optional value from type SMPTE2086
E/IMGMapper(18048): set:488 set: Unset optional value from type CTA861_3
E/IMGMapper(18048): set:488 set: Unset optional value from type SMPTE2086
E/IMGMapper(18048): set:488 set: Unset optional value from type CTA861_3
E/IMGMapper(18048): set:488 set: Unset optional value from type SMPTE2086
E/IMGMapper(18048): set:488 set: Unset optional value from type CTA861_3
E/IMGMapper(18048): set:488 set: Unset optional value from type SMPTE2086
E/IMGMapper(18048): set:488 set: Unset optional value from type CTA861_3
D/BluetoothGatt(18048): discoverServices() - device: D029BA_C
D/BluetoothGatt(18048): onSearchComplete() = Device=D029BA_C Status=0
D/BluetoothGatt(18048): configureMTU() - device: D029BA_C mtu: 517
E/IMGMapper(18048): set:488 set: Unset optional value from type SMPTE2086
E/IMGMapper(18048): set:488 set: Unset optional value from type CTA861_3
D/BluetoothGatt(18048): onConfigureMTU() - Device=D029BA_C mtu=247 status=4