yanshouwang/bluetooth_low_energy #25: Always happening on Android

zhuwei, 2023-11-06:
I execute CentralManager. instance. startDiscovery() in initState, and almost always the following error occurs
```
PlatformException (PlatformException(IllegalStateException, java.lang.IllegalStateException: Start discovery failed with error code: 2, Cause: null, Stacktrace: java.lang.IllegalStateException: Start discovery failed with error code: 2
	at dev.yanshouwang.bluetooth_low_energy_android.MyCentralManager.onScanFailed(MyCentralManager.kt:408)
	at dev.yanshouwang.bluetooth_low_energy_android.MyScanCallback.onScanFailed(MyScanCallback.kt:9)
	at android.bluetooth.le.BluetoothLeScanner$1.run(BluetoothLeScanner.java:568)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:359)
	at android.app.ActivityThread.main(ActivityThread.java:7418)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)
, null))
```

yanshouwang, 2023-11-06:
可以提供一个 demo 吗？ 按照[官方文档](https://developer.android.com/reference/android/bluetooth/le/ScanCallback.html#SCAN_FAILED_APPLICATION_REGISTRATION_FAILED)的说法  errorCode = 2 表示注册扫描回调失败
另外可以试一下项目里的 example 是否存在这个问题

zhuwei, 2023-11-06:
@yanshouwang 我重启了两台android设备又好了。暂时没法再现。。。

yanshouwang, 2023-11-06:
看起来像是手机的蓝牙出了问题，我这边目前没出现过这个现象，如果再出现这个情况的话，麻烦提供下手机型号，系统版本号，还有可复现的代码，方便定位问题

zhuwei, 2023-11-21:
> 看起来像是手机的蓝牙出了问题，我这边目前没出现过这个现象，如果再出现这个情况的话，麻烦提供下手机型号，系统版本号，还有可复现的代码，方便定位问题

Sony Xperia XZ3
不过现在一直没出现了。

yanshouwang, 2023-11-21:
无法复现，看起来是系统问题

closed 2023-11-21 by yanshouwang