yanshouwang/bluetooth_low_energy #27: No named parameter with the name 'hashCode'

chuanpham, 2023-11-09:
Sorry, 
Since the bluetooth_low_energy_platform_interface is updated to verson 4.1.1, this error happens:
`../../.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-4.0.0/lib/src/my_api.dart:87:7: Error: No named parameter with the name 'hashCode'.`

yanshouwang, 2023-11-09:
Sorry about this, the 4.1.1 is not compatible with 4.0.0.
Can you paste your `pubspec.yaml` here?

chuanpham, 2023-11-09:
> Sorry about this, the 4.1.1 is not compatible with 4.0.0. Can you paste your `pubspec.yaml` here?

Here it is, thank you!
```
dependencies:
  chopper: ^6.1.2
  shared_preferences: ^2.0.15
  percent_indicator: ^4.0.1
  share_plus: ^7.0.1
  http: ^1.1.0
  image_picker: ^1.0.4
  cached_network_image: ^3.2.3
  path_provider: ^2.0.11
  screenshot: ^2.1.0
  flutter:
    sdk: flutter

  # The following adds the Cupertino Icons font to your application.
  # Use with the CupertinoIcons class for iOS style icons.
  cupertino_icons: ^1.0.2
  jwt_decode: ^0.3.1
  modal_bottom_sheet: ^2.1.0
  flutter_screenutil: ^5.5.3+2
  fluttertoast: ^8.0.9
  image_cropper: ^5.0.0
  simple_gesture_detector: ^0.2.0
  intl: ^0.18.0
  get: ^4.6.5
  visibility_detector: ^0.4.0+2
  scrollable_positioned_list: ^0.3.4
  flutter_native_splash: ^2.2.9
  flutter_neumorphic: ^3.2.0
  url_launcher: ^6.1.6
  dots_indicator: ^3.0.0
  open_store: ^0.5.0
  package_info_plus: ^4.2.0
  palette_generator: ^0.3.3+2
  fixnum: ^1.0.1
  protobuf: ^2.1.0
  web_socket_channel: ^2.3.0
  freezed_annotation: ^2.2.0
  grpc: ^3.1.0
  jwt_decoder: ^2.0.1
  flutter_slidable: ^3.0.1
  just_the_tooltip: ^0.0.12
  connectivity_plus: ^5.0.1
  time_zone_list: ^1.0.0
  dio: ^5.2.1
  mobkit_dashed_border: ^0.0.5
  nakama:
    git:
      url: git@github.com:chuanpham/nakama-custom.git
      ref: ea37c53
  bluetooth_low_energy: ^4.0.0 <---- HERE

  #service locator
  get_it: ^7.6.0

  #state management
  flutter_bloc: ^8.1.2

  #comparing dart objects
  equatable: ^2.0.5

  #analyze events
  rudder_sdk_flutter: ^2.5.1

  #In-app Review
  in_app_review: ^2.0.6

dev_dependencies:
  flutter_test:
    sdk: flutter

  flutter_lints: ^3.0.1
```

yanshouwang, 2023-11-09:
I can reproduce this error，wait a mintute... Let me think what I shoud do to fix this.

yanshouwang, 2023-11-09:
I have retracted the wrong versions
Please delete your project's `pubspec.lock` file and run `flutter pub get` again, the error shoud fixed after that.

chuanpham, 2023-11-09:
> I have retracted the wrong versions Please delete your project's `pubspec.lock` file and run `flutter pub get` again, the error shoud fixed after that.

Thanks, the error is gone now.

yanshouwang, 2023-11-09:
Fixed

closed 2023-11-09 by yanshouwang

chuanpham, 2023-11-09:
> Fixed

I'm sorry, but after checking again, the problem appears again, please take a look:

`Error (Xcode): ../../.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-4.0.0/lib/src/my_api.dart:87:7: Error: No named parameter with the name 'hashCode'.`

```
class MyGattService2 extends MyGattService {
  late final MyPeripheral peripheral;

  MyGattService2({
    //super.hashCode,
    required super.uuid,
    required List<MyGattCharacteristic2> characteristics,
  }) : super(characteristics: characteristics);

  @override
  List<MyGattCharacteristic2> get characteristics =>
      super.characteristics.cast<MyGattCharacteristic2>();
}
```

```
extension MyGattServiceArgsX on MyGattServiceArgs {
  MyGattService2 toService2() {
    final hashCode = hashCodeArgs;
    final uuid = UUID.fromString(uuidArgs);
    final characteristics = characteristicsArgs
        .cast<MyGattCharacteristicArgs>()
        .map((args) => args.toCharacteristic2())
        .toList();
    return MyGattService2(
      hashCode: hashCode, <---- HERE
      uuid: uuid,
      characteristics: characteristics,
    );
  }
}
```
Thank you for your patience!

tilongzs, 2023-12-28:
I still encountered this problem. Brand new Flutter project and use sample code.

> AppData/Local/Pub/Cache/hosted/mirrors.tuna.tsinghua.edu.cn%2547dart-pub%2547/bluetooth_low_energy_darwin-4.0.0/lib/src/my_api.dart:72:7: Error: No named parameter with the name 'hashCode'.

yanshouwang, 2023-12-28:
> I still encountered this problem. Brand new Flutter project and use sample code.
> 
> > AppData/Local/Pub/Cache/hosted/mirrors.tuna.tsinghua.edu.cn%2547dart-pub%2547/bluetooth_low_energy_darwin-4.0.0/lib/src/my_api.dart:72:7: Error: No named parameter with the name 'hashCode'.

Can you paste the `bluetooth_low_energy_platform_interface` part in your pubspec.lock file, this should caused by the mirror site, I revoked the versions of 4.x.x in [pub.dev](https://pub.dev/packages/bluetooth_low_energy_platform_interface/versions), but I still can found those packages in the [TUNA mirror site](https://mirrors.tuna.tsinghua.edu.cn/dart-pub/packages/bluetooth_low_energy_platform_interface/versions/)

You can use a fixed version in this way
``` Dart
bluetooth_low_energy: 4.0.0
```

Or you can try the pre release version of `5.0.0-dev.3`

tilongzs, 2023-12-28:
Thanks.Solved it in this way.
`bluetooth_low_energy_platform_interface: 4.0.0`