yanshouwang/bluetooth_low_energy #30: 未进行配对，直接发送数据，由于每次最大只能发送500多点，现在速度很慢，有什么办法改善吗？

zhuwei, 2023-11-22:
还是说只能配对后再发送？
centralManager.writeCharacteristic
用的这个发数据

yanshouwang, 2023-11-22:
BLE 不需要配对，如果时无响应写入，通过 getMaximumWriteLength 获取到单次最大写入长度（字节），如果是 20 字节的话速度会比较慢（BLE4.0 蓝牙模块的限制）, 4.2 应该可以协商到 244，5.0 可以协商到 512 字节，我这边用手机测试的话速度可以达到 100kB/s，需要确认下你那边通讯的蓝牙模块的版本，还有最大写入长度是多少

zhuwei, 2023-11-22:
> BLE 不需要配对
> 
> 如果是无响应写入，通过 getMaximumWriteLength 获取到单次最大写入长度（字节），如果是 20 字节的话速度会比较慢（BLE4.0 蓝牙模块的限制）, 4.2 应该可以协商到 244，5.0 可以协商到 512 字节
> 
> 如果是有响应写入，每次最多写入 512 字节，但是比无响应写入会慢（多了 ACK 确认）
> 
> 我这边用手机测试的话无响应写入速度可以达到 100kB/s，需要确认下你那边通讯的蓝牙模块的版本，还有最大写入长度是多少
> 
> PS：蓝牙的连接参数也会影响蓝牙速率，不过 iOS 这边几乎没什么参数可以调整，所以插件里没有提供这方面的接口，用的默认值

我这边用的是两台android设备，我确认每次是512字节。
withoutResponse也就比withResponse快了2倍。最多3-5KB/s.
android设备好像看不出来蓝牙模板版本。蓝牙应用能看到13

zhuwei, 2023-11-22:
我是通过000000c9-0000-1000-8000-00805f9b34fb这个service发送的。你测试是用哪个service发送？

zhuwei, 2023-11-22:
我知道了，要用writeWithoutResponse的Gatt

closed 2023-11-22 by zhuwei

yanshouwang, 2023-11-22:
没太看明白，但是速度问题现在解决了是吧

zhuwei, 2023-11-22:
> 没太看明白，但是速度问题现在解决了是吧

嗯，差不多，无响应写入，能达到20-30KB/S，能凑合了

zhuwei, 2023-11-22:
@yanshouwang 不过无响应或有响应 都存在丢包的情况。。。。

yanshouwang, 2023-11-22:
写个 demo 看下，丢包一般是上层代码问题

zhuwei, 2023-11-22:
> 写个 demo 看下，丢包一般是上层代码问题

我知道了，上层代码问题。非常感谢！