yanshouwang/bluetooth_low_energy #32: Discovery mode misses some devices and shows duplicates on Linux

acfsmith, 2023-12-09:
I tried example on Linux.  After I commented PeripheralManager I got this picture:

[image]

Some devices listed few times, others missing.  Here is screenshot that shows actual devices discovered by OS at the same time:

[image]

yanshouwang, 2023-12-09:
Thanks for your report!

The repeated devices seems like a bug, it will be fixed in the next release.

The missing devices I guess maybe system cache, can you make sure that the missing devices were still advertising at that time?

acfsmith, 2023-12-09:
Yes, I can guarantee that at least one of them (Tacx NEO) was actively advertising at that moment.  I also tested it from MacOS and it appeared in the list as expected.

yanshouwang, 2023-12-10:
> Yes, I can guarantee that at least one of them (Tacx NEO) was actively advertising at that moment. I also tested it from MacOS and it appeared in the list as expected.

I just rememberd that I occured this issue before, seems the [`bluz.dart`](https://github.com/canonical/bluez.dart/tree/main) package can only discover peripherals which advertisement contains [`flag`](https://www.bluetooth.com/blog/advertising-works-part-2/) field with a `General Discoverable Mode` bit mask.

I dont' konw how to make the `bluez` package can discover devices without this flag for now, and the `bluz.dart` package has not updated for a long time, issues even not be answered there, no ideas how to resolve this for now.

I think the only way to make the device to be discovered is add the flag value to the device's advertisement.

acfsmith, 2023-12-10:
I tested the 5.0.0 branch.  Scanning works fine for me now.  Thanks a lot!

yanshouwang, 2023-12-10:
> I tested the 5.0.0 branch. Scanning works fine for me now. Thanks a lot!

Glad to hear that, there are still some work(Android and iOS) to do before the 5.0.0 release, I will make a pre realease version once I finished that.

yanshouwang, 2023-12-27:
The 5.0.0-dev.2 has pre released and should fixed this issue.

acfsmith, 2023-12-27:
Now it can easily detect devices in Linux.  But the discovery callback behaviour has changed.  Before this release, it called discovery callback for every discovered device after every advertisement.  Now it calls discovery callback very rarely, so the signal strength isn't updated.  Also, because of this, I can't tell when the device has finished advertising.

yanshouwang, 2023-12-27:
> Now it can easily detect devices in Linux. But the discovery callback behaviour has changed. Before this release, it called discovery callback for every discovered device after every advertisement. [...]

It's weird but I didn't change the discover logic at all, I tested and I can reproduce this, the RSSI didn't changed after added, but when I opened the system's bluetooth setting pannel, the RSSI will change for some time, after a few moment later the RSSI didn't change again until I reopened the bluetooth setting again...

Curious if this normal in the previous version?

closed 2023-12-30 by yanshouwang