yanshouwang/bluetooth_low_energy #33: Showing duplicate service

chuanpham, 2023-12-11:
I use this plugin to be able to broadcast heart rate via BLE,

Everything works great, but after a while about 10 to 15 minutes after "startAdvertising", I see a second service appear like below:

[image]

I use iOS to advertise service, and scanner is android devices, do you know the reason why? 

Thanks a lot!

yanshouwang, 2023-12-11:
After seached the internet, I think this is a normal behaivor, look at this article: https://novelbits.io/bluetooth-address-privacy-ble/

The iOS used `Resolvable Random Private Address` which will changed about every 15 mintes.

The Android system also use random address, the MAC address changed every time the `startAdvertising` is called.

chuanpham, 2023-12-12:
> After seached the internet, I think this is a normal behaivor, look at this article: https://novelbits.io/bluetooth-address-privacy-ble/
> 
> The iOS used `Resolvable Random Private Address`, the MAC address changed about every 15 mintes.
> 
> The Android system also use random address, the MAC address changed every time the `startAdvertising` is called.

Thank you very much, I got it now

closed 2023-12-12 by yanshouwang