yanshouwang/bluetooth_low_energy #34: Windows Scanner doesn't show devices names

iBog, 2023-12-29:
[image]

yanshouwang, 2023-12-30:
This is because the peripheral's advertisement didn't contains a local name field, we use this name when discover advertisement, you can try the [BluetoothLEAdvertisement](https://github.com/microsoft/Windows-universal-samples/tree/main/Samples/BluetoothAdvertisement) sample of Microsoft, it's should have the same behavior(you should comment the watcher's filters code first)
If the advertisement contains a local name field, you will see the name when run the example, you can try to run this example with an Android or iOS device which support PeripheralManager api, use the advertiser to advertise an advertisement, then run this example on Windows or other platforms, you can see the name(by default le12138) after scan.
[image]

yanshouwang, 2024-03-15:
Thanks for your report, let me check the code to see how to resolve this issue.

yanshouwang, 2024-04-07:
> I had this happen to me and like you mentioned the example works.
>
> After some playing around with it, I believe there is a parsing issue. Names with non alpha-numeric characters don't show up.
>
> Works unclebob
>
> doesn't work uncle bob (also tried variant uncle_bob which didn't work)

I tested on windows but can't reproduce this issue.

[image]

[image]

[image]

sampaioletti, 2024-05-08:
I'm still planning on looking into this further just haven't had time

yanshouwang, 2024-05-08:
> I'm still planning on looking into this further just haven't had time

I tested this a few days before, I believe this is a normal behaivor, you can run the [Microsoft samples](https://github.com/microsoft/Windows-universal-samples/tree/main/Samples/BluetoothAdvertisement/cs), it has the same behavior as this plugin.

Notice there are two ways to discover BLE devices on Windows, one way is to use the [`BluetoothLEAdvertisementWatcher`](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisementwatcher?view=winrt-22621), another way is to use the [`DeviceWatcher`](https://learn.microsoft.com/en-us/uwp/api/windows.devices.enumeration.devicewatcher?view=winrt-22621) class.

The first way which this plugin used can only show the localName defined in the BLE advertisement.

The second way does show the device name but this name is not defined in the BLE advertisement, and use this way we can't get the BLE advertisement and RSSI.

PS:
The first way sample from Microsoft is [here](https://github.com/microsoft/Windows-universal-samples/tree/main/Samples/BluetoothAdvertisement/cs). You need to comment the Advertisement filters before run this sample.

The second way sample from Microsoft is [here](https://github.com/microsoft/Windows-universal-samples/tree/main/Samples/BluetoothLE/cs).

yanshouwang, 2024-05-22:
Good news! I have confirmed that the missed device name is included in the scan response or extended advertisement, which I ignored before, this will be fixed in the next version. Anyone can pull the 6.0.0 branch and run the example in the bluetooth_low_energy_windows to see the difference.

sampaioletti, 2024-05-22:
Awesome I'll test that out...sorry I never had time to pursue. Thanks for the great lib.

yanshouwang, 2024-06-03:
Fixed in 6.0.0

closed 2024-06-03 by yanshouwang