yanshouwang/bluetooth_low_energy #36: (iOS) Not able to discover GATT Characteristics

IvanTorres21, 2024-01-12:
Hi!

I've compiled the example project (Version 5.0.0), it works on Android and Windows, but when compiling for iOS it never discovers the services and GATT characteristics. 

Steps to reproduce:

1. Download the example project
2. Run `flutter pub get`
3. Run `flutter run`
4. Scan devices
5. Connect to device

And the Services list will never get values.

yanshouwang, 2024-01-12:
I tested on macOS(My apple developer account was unregistered so I can't test on my iPhone device right now) and connect the Gatt server on my Android device, can't reproduce this issue, but iOS and macOS use the same code, can you run the GattServer on a Android device and test again to see if this issue still exists? and provide the log messages on the console will be helpful.
[image]

yanshouwang, 2024-01-15:
Tested on iPhone but can't reproduce this issue.

IvanTorres21, 2024-01-15:
Sorry for the delay, I didn't have access to the project during the weekend.

Here is the iPhone log.

```
App crached.

PlatformException(illegalArgument, MyError, Stacktrace: ["0   bluetooth_low_energy_darwin         0x00000001051b2f0c $s27bluetooth_low_energy_darwin9wrapError33_67DD5082666B3ED7A98ED592AD08C18CLLySayypSgGypF + 1300", "1   bluetooth_low_energy_darwin         0x00000001051bc8ac $s27bluetooth_low_energy_darwin28MyCentralManagerHostApiSetupC5setUp15binaryMessenger3apiySo013FlutterBinaryN0_p_AA0efghI0_pSgtFZyypSg_yAJctcfU8_ys6ResultOySayAA0E18GattDescriptorArgsVGs5Error_pGcfU_ + 384", "2   bluetooth_low_energy_darwin         0x00000001051d3ae4 $s27bluetooth_low_energy_darwin16MyCentralManagerC19discoverDescriptors8uuidArgs08hashCodeK010completionySS_s5Int64Vys6ResultOySayAA0e14GattDescriptorK0VGs5Error_pGctF + 1020", "3   bluetooth_low_energy_darwin         0x00000001051e0588 $s27bluetooth_low_energy_darwin16MyCentralManagerCAA0efG7HostApiA2aDP19discoverDescriptors8uuidArgs08hashCodeM010completionySS_s5Int64Vys6ResultOySayAA0e14GattDescriptorM0VGs5Error_pGctFTW + 28", "4   bluetooth_low_energy_darwin         0x00000001051bc704 $s27bluetooth_low_energy_darwin28MyCentralManagerHostApiSetupC5setUp15binaryMessenger3apiySo013FlutterBinaryN0_p_AA0efghI0_pSgtFZyypSg_yAJctcfU8_ + 1240", "5   bluetooth_low_energy_darwin         0x00000001051b9ed4 $sypSgAAIegn_Iegng_yXlSgABIeyBy_IeyByy_TR + 236", "6   Flutter                             0x0000000108547914 __48-[FlutterBasicMessageChannel setMessageHandler:]_block_invoke + 160", "7   Flutter                             0x0000000107fa1644 ___ZN7flutter25PlatformMessageHandlerIos21HandlePlatformMessageENSt21_LIBCPP_ABI_NAMESPACE10unique_ptrINS_15PlatformMessageENS1_14default_deleteIS3_EEEE_block_invoke + 116", "8   libdispatch.dylib                   0x00000001c86207a8 30AB7A30-04BC-3C9C-AE24-8DE17D9D2AF2 + 407464", "9   libdispatch.dylib                   0x00000001c8621780 30AB7A30-04BC-3C9C-AE24-8DE17D9D2AF2 + 411520", "10  libdispatch.dylib                   0x00000001c85cdbdc 30AB7A30-04BC-3C9C-AE24-8DE17D9D2AF2 + 68572", "11  libdispatch.dylib                   0x00000001c85cd82c 30AB7A30-04BC-3C9C-AE24-8DE17D9D2AF2 + 67628", "12  CoreFoundation                      0x00000001c19059ac 06CFD2AC-08FA-3879-8894-C29869F8CD97 + 596396", "13  CoreFoundation                      0x00000001c18e9648 06CFD2AC-08FA-3879-8894-C29869F8CD97 + 480840", "14  CoreFoundation                      0x00000001c18edd20 CFRunLoopRunSpecific + 584", "15  GraphicsServices                    0x00000001f8478998 GSEventRunModal + 160", "16  UIKitCore                           0x00000001c3b7f80c 11874256-4FD7-3E33-8914-E04D909E9387 + 3610636", "17  UIKitCore                           0x00000001c3b7f484 UIApplicationMain + 312", "18  Runner                              0x0000000104f7459c main + 64", "19  dyld                                0x00000001df098344 9808DB40-10F3-3EC8-8E14-3A370A6619E7 + 82756"], null)

#0      MyCentralManagerHostApi.discoverDescriptors (package:bluetooth_low_energy_darwin/src/my_api.g.dart:548:7)
<asynchronous suspension>
#1      MyCentralManager.discoverGATT.<anonymous closure> (package:bluetooth_low_energy_darwin/src/my_central_manager.dart:127:19)
<asynchronous suspension>
#2      MyCentralManager.discoverGATT (package:bluetooth_low_energy_darwin/src/my_central_manager.dart:125:33)
<asynchronous suspension>
#3      _PeripheralViewState.buildAppBar.<anonymous closure>.<anonymous closure> (package:bluetooth_low_energy_example/main.dart:489:23)
<asynchronous suspension>
``` 

The app is still open and I can move between the scan and connect menu without problem, but the services won't show (though the logs kind of look like it was able to read some?) 

Also opened the Android version to show you that it's not a problem with the device  here is screenshot.

[image]

yanshouwang, 2024-01-15:
@IvanTorres21 Can you provide the `pubspec.yaml` and `pubspec.lock` of your project? I guess this may be a wrong dependency problem.

IvanTorres21, 2024-01-16:
Here they are. 

Pubseck.yaml

```
name: bluetooth_low_energy_example
description: Demonstrates how to use the bluetooth_low_energy plugin.
# The following line prevents the package from being accidentally published to
# pub.dev using `flutter pub publish`. This is preferred for private packages.
publish_to: 'none' # Remove this line if you wish to publish to pub.dev
​
environment:
  sdk: '>=3.0.0 <4.0.0'
​
# Dependencies specify other packages that your package needs in order to work.
# To automatically upgrade your package dependencies to the latest versions
# consider running `flutter pub upgrade --major-versions`. Alternatively,
# dependencies can be manually updated by changing the version numbers below to
# the latest version available on pub.dev. To see which dependencies have newer
# versions available, run `flutter pub outdated`.
dependencies:
  flutter:
    sdk: flutter
​
  bluetooth_low_energy: ^5.0.1
​
  # The following adds the Cupertino Icons font to your application.
  # Use with the CupertinoIcons class for iOS style icons.
  cupertino_icons: ^1.0.2
  convert: ^3.1.1
  intl: ^0.19.0
​
dev_dependencies:
  integration_test:
    sdk: flutter
  flutter_test:
    sdk: flutter
​
  # The "flutter_lints" package below contains a set of recommended lints to
  # encourage good coding practices. The lint set provided by the package is
  # activated in the `analysis_options.yaml` file located at the root of your
  # package. See that file for information about deactivating specific lint
  # rules and activating additional ones.
  flutter_lints: ^3.0.0
​
# For information on the generic Dart part of this file, see the
# following page: https://dart.dev/tools/pub/pubspec
​
# The following section is specific to Flutter packages.
flutter:
​
  # The following line ensures that the Material Icons font is
  # included with your application, so that you can use the icons in
  # the material Icons class.
  uses-material-design: true
​
  # To add assets to your application, add an assets section, like this:
  # assets:
  #   - images/a_dot_burr.jpeg
  #   - images/a_dot_ham.jpeg
​
  # An image asset can refer to one or more resolution-specific "variants", see
  # https://flutter.dev/assets-and-images/#resolution-aware
​
  # For details regarding adding assets from package dependencies, see
  # https://flutter.dev/assets-and-images/#from-packages
​
  # To add custom fonts to your application, add a fonts section here,
  # in this "flutter" section. Each entry in this list should have a
  # "family" key with the font family name, and a "fonts" key with a
[... 393 lines cut ...]
      sha256: afe077240a270dcfd2aafe77602b4113645af95d0ad31128cc02bce5ac5d5152
      url: "https://pub.dev"
    source: hosted
    version: "0.3.0"
  webdriver:
    dependency: transitive
    description:
      name: webdriver
      sha256: "3c923e918918feeb90c4c9fdf1fe39220fa4c0e8e2c0fffaded174498ef86c49"
      url: "https://pub.dev"
    source: hosted
    version: "3.0.2"
  xml:
    dependency: transitive
    description:
      name: xml
      sha256: b015a8ad1c488f66851d762d3090a21c600e479dc75e68328c52774040cf9226
      url: "https://pub.dev"
    source: hosted
    version: "6.5.0"
sdks:
  dart: ">=3.2.0 <4.0.0"
  flutter: ">=3.3.0"

```

closed 2024-01-16 by yanshouwang