yanshouwang/bluetooth_low_energy #41: Unhandled Exception: PlatformException(NullPointerException, java.lang.NullPointerException: manager.openGattServer(m…etoothGattServerCallback) must not be null, Cause: null, Stacktrace: java.lang.NullPointerException: manager.openGattServer(m…etoothGattServerCallback) must not be null

jcm10x1, 2024-01-18:
I am getting this error on app startup. It worked once, then stopped.
```
I/Choreographer(23721): Skipped 604 frames!  The application may be doing too much work on its main thread.
E/flutter (23721): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(NullPointerException, java.lang.NullPointerException: manager.openGattServer(m…etoothGattServerCallback) must not be null, Cause: null, Stacktrace: java.lang.NullPointerException: manager.openGattServer(m…etoothGattServerCallback) must not be null
E/flutter (23721):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManager.mOpenServer(MyPeripheralManager.kt:456)
E/flutter (23721):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManager.mOnStateChanged(MyPeripheralManager.kt:447)
E/flutter (23721):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManager.setUp(MyPeripheralManager.kt:106)
E/flutter (23721):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManagerHostApi$Companion.setUp$lambda-1$lambda-0(MyApi.g.kt:821)
E/flutter (23721):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManagerHostApi$Companion.$r8$lambda$rPfT_nG-ruq9HW1JuLDRUvvp9Hw(Unknown Source:0)
E/flutter (23721):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManagerHostApi$Companion$$ExternalSyntheticLambda0.onMessage(Unknown Source:2)
E/flutter (23721):      at io.flutter.plugin.common.BasicMessageChannel$IncomingMessageHandler.onMessage(BasicMessageChannel.java:261)
E/flutter (23721):      at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/flutter (23721):      at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
E/flutter (23721):      at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/flutter (23721):      at android.os.Handler.handleCallback(Handler.java:883)
E/flutter (23721):      at android.os.Handler.dispatchMessage(Handler.java:100)
E/flutter (23721):      at android.os.Looper.loop(Looper.java:214)
E/flutter (23721):      at android.app.ActivityThread.main(ActivityThread.java:7615)
E/flutter (23721):      at java.lang.reflect.Method.invoke(Native Method)
E/flutter (23721):      at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
E/flutter (23721):      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)
E/flutter (23721): , null)
E/flutter (23721): #0      MyPeripheralManagerHostApi.setUp (package:bluetooth_low_energy_android/src/my_api.g.dart:887:7)
E/flutter (23721): <asynchronous suspension>
E/flutter (23721): #1      MyPeripheralManager.setUp (package:bluetooth_low_energy_android/src/my_peripheral_manager.dart:63:5)
E/flutter (23721): <asynchronous suspension>
E/flutter (23721): #2      main (package:e_drum/main.dart:25:3)
E/flutter (23721): <asynchronous suspension>
```

yanshouwang, 2024-01-18:
Can you test the example to see if this error exists? Can you make a minimal project for me to reproduce this issue?

jcm10x1, 2024-01-18:
@yanshouwang The example has the same error. Is the example an ok minimum reproduction? Here is the entire log.
```
The Flutter DevTools debugger and profiler on LM Q730 is available at:
http://127.0.0.1:9103?uri=http://127.0.0.1:56142/q9XiQQqccrM=/
D/BluetoothGattServer(27985): registerCallback()
D/BluetoothGattServer(27985): registerCallback() - UUID=4058a749-38c9-41c5-b875-10c99b602a51
I/Choreographer(27985): Skipped 604 frames!  The application may be doing too much work on its main thread.
D/Surface (27985): Surface::disconnect(this=0x7e0ef80000,api=1)
D/Surface (27985): Surface::connect(this=0x7e0ef80000,api=1)
D/Surface (27985): Surface::setBufferCount(this=0x7e0ef80000,bufferCount=3)
D/Surface (27985): Surface::disconnect(this=0x7e17331000,api=1)
D/Surface (27985): Surface::connect(this=0x7e17331000,api=1)
D/Surface (27985): Surface::disconnect(this=0x7e0ef80000,api=1)
D/Surface (27985): Surface::connect(this=0x7e0ef80000,api=1)
D/Surface (27985): Surface::setBufferCount(this=0x7e0ef80000,bufferCount=3)
D/Surface (27985): Surface::disconnect(this=0x7e0ef80000,api=1)
D/Surface (27985): Surface::connect(this=0x7e0ef80000,api=1)
D/Surface (27985): Surface::setBufferCount(this=0x7e0ef80000,bufferCount=3)
D/Surface (27985): Surface::disconnect(this=0x7e17331000,api=1)
D/Surface (27985): Surface::connect(this=0x7e17331000,api=1)
D/Surface (27985): Surface::disconnect(this=0x7e0ef80000,api=1)
D/Surface (27985): Surface::connect(this=0x7e0ef80000,api=1)
D/Surface (27985): Surface::setBufferCount(this=0x7e0ef80000,bufferCount=3)

Performing hot restart...                                               
Restarted application in 2,605ms.
D/BluetoothGattServer(27985): registerCallback()
D/BluetoothGattServer(27985): registerCallback() - UUID=9c1327ce-31b7-44f0-9965-fe383a2c213a
I/Choreographer(27985): Skipped 602 frames!  The application may be doing too much work on its main thread.
I/GED     (27985): ged_boost_gpu_freq, level 100, eOrigin 2, final_idx 3, oppidx_max 3, oppidx_min 0
D/Surface (27985): Surface::disconnect(this=0x7e17331000,api=1)
D/Surface (27985): Surface::disconnect(this=0x7e17331000,api=-1)
D/Surface (27985): Surface::disconnect(this=0x7e0ef80000,api=1)
I/GED     (27985): ged_boost_gpu_freq, level 100, eOrigin 2, final_idx 3, oppidx_max 3, oppidx_min 0
I/Timeline(27985): Timeline: Activity_idle id: android.os.BinderProxy@dd4d2c0 time:1130514448
I/Timeline(27985): Timeline: Activity_idle id: android.os.BinderProxy@dd4d2c0 time:1130514450
D/Surface (27985): Surface::connect(this=0x7e17330000,api=1)
D/Surface (27985): Surface::setBufferCount(this=0x7e17330000,bufferCount=3)
D/Surface (27985): Surface::connect(this=0x7e0ef28000,api=1)
D/BluetoothGattServer(27985): onServerRegistered() - status=0 serverIf=7
E/BluetoothGattServer(27985): onServerRegistered: mCallback is null
D/BluetoothGattServer(27985): onServerRegistered() - status=0 serverIf=12
E/BluetoothGattServer(27985): onServerRegistered: mCallback is null
I/GED     (27985): ged_boost_gpu_freq, level 100, eOrigin 2, final_idx 3, oppidx_max 3, oppidx_min 0
D/Surface (27985): Surface::disconnect(this=0x7e0ef28000,api=1)
D/Surface (27985): Surface::disconnect(this=0x7e0ef28000,api=-1)
D/Surface (27985): Surface::disconnect(this=0x7e17330000,api=1)
I/GED     (27985): ged_boost_gpu_freq, level 100, eOrigin 2, final_idx 3, oppidx_max 3, oppidx_min 0
I/Timeline(27985): Timeline: Activity_idle id: android.os.BinderProxy@dd4d2c0 time:1130577133
E/flutter (24872): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(NullPointerException, java.lang.NullPointerException: manager.openGattServer(m…etoothGattServerCallback) must not be null, Cause: null, Stacktrace: java.lang.NullPointerException: manager.openGattServer(m…etoothGattServerCallback) must not be null
E/flutter (24872):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManager.mOpenServer(MyPeripheralManager.kt:456)
E/flutter (24872):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManager.mOnStateChanged(MyPeripheralManager.kt:447)
E/flutter (24872):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManager.setUp(MyPeripheralManager.kt:106)
E/flutter (24872):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManagerHostApi$Companion.setUp$lambda-1$lambda-0(MyApi.g.kt:821)
E/flutter (24872):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManagerHostApi$Companion.$r8$lambda$rPfT_nG-ruq9HW1JuLDRUvvp9Hw(Unknown Source:0)
E/flutter (24872):      at dev.yanshouwang.bluetooth_low_energy_android.MyPeripheralManagerHostApi$Companion$$ExternalSyntheticLambda0.onMessage(Unknown Source:2)
E/flutter (24872):      at io.flutter.plugin.common.BasicMessageChannel$IncomingMessageHandler.onMessage(BasicMessageChannel.java:261)
E/flutter (24872):      at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/flutter (24872):      at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
E/flutter (24872):      at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/flutter (24872):      at android.os.Handler.handleCallback(Handler.java:883)
E/flutter (24872):      at android.os.Handler.dispatchMessage(Handler.java:100)
E/flutter (24872):      at android.os.Looper.loop(Looper.java:214)
[... cut ...]
E/flutter (24872): , null)
E/flutter (24872): #0      MyPeripheralManagerHostApi.setUp (package:bluetooth_low_energy_android/src/my_api.g.dart:887:7)
E/flutter (24872): <asynchronous suspension>
E/flutter (24872): #1      MyPeripheralManager.setUp (package:bluetooth_low_energy_android/src/my_peripheral_manager.dart:63:5)
E/flutter (24872): <asynchronous suspension>
E/flutter (24872): #2      main (package:e_drum/main.dart:25:3)
E/flutter (24872): <asynchronous suspension>
E/flutter (24872): 
```

yanshouwang, 2024-01-18:
@jcm10x1 It's wired, I can't reproduce this with the example, here is the log
``` 
Launching lib\main.dart on 23013RK75C in debug mode...
√  Built build\app\outputs\flutter-apk\app-debug.apk.
Connecting to VM Service at ws://127.0.0.1:63120/SHuSzq-sTjc=/ws
D/BluetoothGattServer(13456): registerCallback()
D/BluetoothGattServer(13456): registerCallback() - UUID=17f2f6ad-98ec-4590-8cc3-edbae3f5ce69
D/BluetoothGattServer(13456): onServerRegistered() - status=0 serverIf=9
[MyCentralManager] setUp
[MyPeripheralManager] setUp
[MyCentralManager] onStateChanged: MyBluetoothLowEnergyStateArgs.on
[MyPeripheralManager] onStateChanged: MyBluetoothLowEnergyStateArgs.on
[MyCentralManager] getState
E/libEGL  (13456): pre_cache appList: ,,
E/LB      (13456): fail to open file: No such file or directory
W/1.raster(13456): type=1400 audit(0.0:8505): avc: denied { getattr } for path="/sys/module/metis/parameters/minor_window_app" dev="sysfs" ino=69262 scontext=u:r:untrusted_app:s0:c27,c257,c512,c768 tcontext=u:object_r:sysfs_migt:s0 tclass=file permissive=0 app=dev.yanshouwang.bluetooth_low_energy_example
D/_energy_example(13456): MiuiProcessManagerServiceStub setSchedFifo
I/MiuiProcessManagerImpl(13456): setSchedFifo pid:13456, mode:3
I/_energy_example(13456): ProcessProfilingInfo new_methods=0 is saved saved_to_disk=0 resolve_classes_delay=8000
W/Parcel  (13456): Expecting binder but got null!
D/DecorView[](13456): onWindowFocusChanged hasWindowFocus true
I/HandWritingStubImpl(13456): refreshLastKeyboardType: 1
2
I/HandWritingStubImpl(13456): getCurrentKeyboardType: 1
```
What's the Android version of your device? Any other steps to run into this?

yanshouwang, 2024-01-18:
Form the error itself, seems the openGattServer returns null, that's will caused by BLE is powred off or run without UI thread, but the example should work.

jcm10x1, 2024-01-18:
which Android version are you on? Did you see what happens after a hot restart?

yanshouwang, 2024-01-18:
> which Android version are you on? Did you see what happens after a hot restart?

I'm running on an Android 13 Redmi device, the hot restart is also workd
```
Restarted application in 1,279ms.
I/_energy_example(13457): This is non sticky GC, maxfree is 8388608 minfree is 2097152
W/FinalizerDaemon(13457): type=1400 audit(0.0:8594): avc: denied { getopt } for path="/dev/socket/usap_pool_primary" scontext=u:r:untrusted_app:s0:c27,c257,c512,c768 tcontext=u:r:zygote:s0 tclass=unix_stream_socket permissive=0 app=dev.yanshouwang.bluetooth_low_energy_example
[MyCentralManager] setUp
[MyPeripheralManager] setUp
[MyCentralManager] onStateChanged: MyBluetoothLowEnergyStateArgs.on
[MyPeripheralManager] onStateChanged: MyBluetoothLowEnergyStateArgs.on
[MyCentralManager] getState
D/DecorView[](13457): onWindowFocusChanged hasWindowFocus false
```

jcm10x1, 2024-01-18:
I tried on an Android 5 device that I have and it seems to be working based on basic testing. Do you have a way to test on Android 10?

yanshouwang, 2024-01-18:
> I tried on an Android 5 device that I have and it seems to be working based on basic testing. Do you have a way to test on Android 10?

I don't have a Android 10 device right now. If the Android 5 works, maybe this a manufacturer issue? It's like you run into this with a LG devie, you can test this with other brands.

yanshouwang, 2024-01-18:
useful link: https://stackoverflow.com/questions/22981669/why-does-bluetoothmanager-opengattserver-always-return-null-on-my-nexus-7

closed 2024-03-04 by github-actions[bot]