yanshouwang/bluetooth_low_energy #42: Read function on windows always return the initial value

Tunahan0155, 2024-01-18:
Hi, When the notify feature is disabled in Windows, the read button always gives the value when the project is first opened, and then it continues to give the same value even if the value in the character is changed on the peripheral device. This problem does not occur when notify is enabled, it occurs when it is disabled. I did not encounter this problem on Android, iOS, or Mac only on Windows.

yanshouwang, 2024-01-19:
Thanks for this reply, the windows API is a little different from other platforms, it cached values of the peripherals, after looked at the [GattCharacteristic.ReadValueAsync](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattcharacteristic.readvalueasync?view=winrt-22621) document, I need to add the `cacheMode` argument when I call this method.

yanshouwang, 2024-01-19:
@Tunahan0155 Please test with 5.0.5 to see if this error has resolved.

closed 2024-01-19 by yanshouwang

Tunahan0155, 2024-01-19:
Thank you for your answer, I tested it with version 5.0.5, the error was solved.