yanshouwang/bluetooth_low_energy #44: Not working on MacOS

martin-robert-fink, 2024-01-26:
I added the key:

`	<key>NSBluetoothAlwaysUsageDescription</key>
	<string>App uses bluetooth</string>`

to the MacOS info.plist and I only get:

`[MyCentralManager] setUp
[MyCentralManager] onStateChanged: MyBluetoothLowEnergyStateArgs.unsupported
[MyCentralManager] getState`

I've done the same thing for iOS and everything works fine.  Is there something else missing to get the plug in to work on MacOS?

Martin

yanshouwang, 2024-01-26:
Yes, when use bluetooth or other hardware on macOS, developers need to configure the app sandbox, look at this document: https://developer.apple.com/documentation/xcode/configuring-the-macos-app-sandbox#Enable-access-to-restricted-resources

Also the README should update about this, thanks for your report.

martin-robert-fink, 2024-01-26:
Perfect.  That worked, thanks.
Martin

closed 2024-01-26 by martin-robert-fink