yanshouwang/bluetooth_low_energy #49: iOS and MacOS Swift Compiler errors

klipsedeville, 2024-01-31:
I am getting build errors when trying to build for iOS and MacOS. Works great on android. See the errors below. My xcode swift compiler version is set to 5.

Launching lib/main.dart on iPhone 13 in debug mode...
Xcode build done.                                           16.4s
Failed to build iOS app
Uncategorized (Xcode): Command CompileSwift failed with a nonzero exit code
Swift Compiler Error (Xcode): Constrained extension must be declared on the unspecialized generic type 'Array' with constraints specified by a 'where' clause
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:19:0
Swift Compiler Error (Xcode): Constrained extension must be declared on the unspecialized generic type 'Dictionary' with constraints specified by a 'where' clause
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:214:0

Swift Compiler Error (Xcode): Type 'Element' has no member 'read'
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:24:18

Swift Compiler Error (Xcode): Type 'Element' has no member 'write'
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:26:18

Swift Compiler Error (Xcode): Type 'Element' has no member 'writeWithoutResponse'
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:28:18
Swift Compiler Error (Xcode): Type 'Element' has no member 'notify'
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:30:18

Swift Compiler Error (Xcode): Type 'Element' has no member 'indicate'
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:32:18

Swift Compiler Error (Xcode): Type 'Element' has no member 'read'
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:43:18

Swift Compiler Error (Xcode): Type 'Element' has no member 'write'
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:45:18
Swift Compiler Error (Xcode): Type 'Element' has no member 'writeWithoutResponse'
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:45:26

Swift Compiler Error (Xcode): No exact matches in call to subscript 
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:216:27

Swift Compiler Error (Xcode): No exact matches in call to subscript 
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:217:31
Swift Compiler Error (Xcode): No exact matches in call to subscript 
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:219:30

Swift Compiler Error (Xcode): Unable to infer complex closure return type; add explicit type to disambiguate
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:220:59
Swift Compiler Error (Xcode): No exact matches in call to subscript 
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyApi.swift:226:43
Swift Compiler Error (Xcode): Cannot convert return expression of type '[Int64 : CBService]' to return type 'String?'
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyCentralManager.swift:391:23

Swift Compiler Error (Xcode): 'flatMap' has been renamed to 'compactMap(_:)': Please use compactMap(_:) for the case where closure returns an optional value
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyCentralManager.swift:389:34
Swift Compiler Error (Xcode): Type of expression is ambiguous without more context
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyCentralManager.swift:393:32

Swift Compiler Error (Xcode): Cannot convert return expression of type '[Int64 : CBCharacteristic]' to return type 'String?'
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyCentralManager.swift:411:23
Swift Compiler Error (Xcode): 'flatMap' has been renamed to 'compactMap(_:)': Please use compactMap(_:) for the case where closure returns an optional value
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyCentralManager.swift:409:41

Swift Compiler Error (Xcode): Type of expression is ambiguous without more context
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyCentralManager.swift:413:53
Swift Compiler Error (Xcode): Cannot convert return expression of type '[Int64 : CBDescriptor]' to return type 'String?'
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyCentralManager.swift:431:23

Swift Compiler Error (Xcode): 'flatMap' has been renamed to 'compactMap(_:)': Please use compactMap(_:) for the case where closure returns an optional value
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyCentralManager.swift:429:37
Swift Compiler Error (Xcode): Type of expression is ambiguous without more context
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyCentralManager.swift:433:49

Swift Compiler Error (Xcode): Unable to infer complex closure return type; add explicit type to disambiguate
/Users/ikenna/.pub-cache/hosted/pub.dev/bluetooth_low_energy_darwin-5.0.3/darwin/Classes/MyPeripheralManager.swift:184:46
Could not build the application for the simulator.
Error launching application on iPhone 13.
Exited

yanshouwang, 2024-01-31:
@klipsedeville I think you can fix this by upgrade swift version to 5.7 or above
I used the latest Xcode when develope this plugin, so there maybe exist some syntax errors when build with old version Xcode.

See: https://github.com/apple/swift-evolution/blob/main/proposals/0361-bound-generic-extensions.md

klipsedeville, 2024-01-31:
> @klipsedeville I think you can fix this by upgrade swift version to 5.7 or above I used the latest Xcode when develope this plugin, so there maybe exist some syntax errors when build with old [...]

Thanks so much for replying. I will upgrade swift and see if the errors go away.