yanshouwang/bluetooth_low_energy #52: May not be an issue: Connection requesting PIN

klipsedeville, 2024-02-02:
When connecting 2 android phones (not tested on iOS yet), the devices show a system pair and connect dialog. When I click the pair option and confirm on the other device, it shows and error that the PIN is invalid.

Is it possible to connect devices, discover characteristics and read/write between them without pairing or having the system dialog pop up? I want to achieve this with zero user involvement, the devices silently connect to each other and exchange data.

yanshouwang, 2024-02-03:
BLE doesn't need to pair when connect to other devices, and this plugin doesn't provide pair method, If you see the pair dialog when connect, you should check your peripheral side to change the connection mode to  [Just works](https://stackoverflow.com/questions/52100749/ble-paring-without-bonding-on-android)

klipsedeville, 2024-02-03:
> BLE can communicate with other devices without bonding, and this plugin doesn't provide bind method, If you see the pair dialog when connect, you should check your peripheral side to change the connection mode to [Just works](https://stackoverflow.com/questions/52100749/ble-paring-without-bonding-on-android)

Thanks for the reply. How do I set the peripheral android device side to the Just works connection mode using your plugin?

yanshouwang, 2024-02-04:
> Thanks for the reply. How do I set the peripheral android device side to the Just works connection mode using your plugin?

There is nothing to do with this plugin，I'm sure I can communicate between central and peripheral use the example without pair dialog.