yanshouwang/bluetooth_low_energy #53: Service UUIDs not in advertisement

martin-robert-fink, 2024-02-22:
Hi - I had to take a little break from this, but had a bit of time today to continue testing/trying.

I'm using a heart rate monitor to try out the plug in.

When scanning, the advertisement data should have list of service UUIDs.  My heart rate monitor should come back with 4 of them.  Heart Rate, Battery, Device Information, and then an unidentified UUID string.  I confirm this using LightBlue.

I have looked at this using my own code, but also using the example code provided with the plugin.  Even with the example app, no service UUIDs show up for the heart rate monitor.  To be clear, the heart rate monitor is scanned/found.  However the service UUID list is empty.

I think this is a bug.

As an aside, you example app gets a linting error with the latest flutter in main.dart on line 282.  Don't use BuildContext across async gaps.  You may want to look at that.

Thanks,
Martin

yanshouwang, 2024-03-14:
Sorry for response this late, too busy these days with other stuff. Because I don't have a heart rate monitor device, I can't debug the code, maybe there are some errors when convert the UUID. Maybe you can debug your app to see the raw data in the Xcode to get the raw data of the advertisement?

martin-robert-fink, 2024-03-14:
I decided to try flutter_blue_plus and it works the way I'd expect. In the scan api, you specify either or both of the withName: and withUuid: parameters and scan returns only those devices that meet the criteria.

yanshouwang, 2024-03-14:
> I decided to try flutter_blue_plus and it works the way I'd expect. In the scan api, you specify either or both of the withName: and withUuid: parameters and scan returns only those devices that meet the criteria.

Yes, the BLE should able to define scan filter when scan devices, I'll add this in the next version.
While the advertisement data seems still need some investigation.

yanshouwang, 2024-05-16:
The `serviceUUIDs` filter is added in the [6.0.0-dev.0](https://pub.dev/packages/bluetooth_low_energy/versions/6.0.0-dev.0)

yanshouwang, 2024-06-03:
Fixed in 6.0.0

closed 2024-06-03 by yanshouwang