yanshouwang/bluetooth_low_energy #54: BLE disconnect on Android issue

iBog, 2024-03-13:
Trying to connect to BLE "HR monitor" device.
App should auto search all devices, found by name, connect, subscribe characteristic to notify every data change.
await CentralManager.instance.setCharacteristicNotifyState(
                characteristic,
                state: true,
              );
All this steps working fine, data start flowing.

On app close (hide) - App should auto disconnect from device.
On this step I call two actions - 
1) await CentralManager.instance.setCharacteristicNotifyState(
                characteristic,
                state: false,
              );
To stop characteristic subscription
2) await CentralManager.instance.disconnect(peripheral);
to disconnect from BLE device itself

After application Resume, need to start from the beginning... Search, Connect... but
BLE device not listed in search list anymore.
As I understand, connection with BLE not stopped, and device still attached.
If temporary switch OFF bluetooth on Android, and run Search BLE devices again - "HR monitor" appeared again, 
because "pairing" was interrupted and device is ready to be searched and connected

Guide me what i did wrong, or package "bluetooth_low_energy" has an issue when App using CharacteristicNotify

iBog, 2024-03-13:
small addition, BLE device is not switched off, it always on me and working,
flutter app should periodically connect to device and disconnect from it.

yanshouwang, 2024-03-14:
Does the connection closed after you called `disconnect` method?

In my opinion, it's the system's duty to disconnect the BLE device when app closed, and our BLE device did disconnected when I close my app.

I found there some cases on the internet that the connection still connected after app closed, but I can't find anything useful...
https://stackoverflow.com/questions/73107781/devices-with-android-12-keep-bluetooth-le-connection-even-when-app-is-closed
https://github.com/evothings/cordova-ble/issues/136

iBog, 2024-03-14:
@yanshouwang according to connectionStateChanged, connectionState become false.
Agree with you, looks like system delay on releasing connection with device.
Also found solution in quick.blue library: https://github.com/pauldemarco/flutter_blue/issues/317#issuecomment-522134619

manjarqo, 2024-12-20:
I have the same question as the topic starter. In version 6.0.2, I am facing the same problem. Does anyone know how to solve it?

yanshouwang, 2024-12-21:
> I have the same question as the topic starter. In version 6.0.2, I am facing the same problem. Does anyone know how to solve it?

There is another possibility on the peripheral side which the peripheral does not handle the disconnect event when app closed. So it will not start advertising again. Then we can not discover it any more. But the connection already closed indeed.