yanshouwang/bluetooth_low_energy #61: Connect to devices already connected to the OS

AYM1607, 2024-04-09:
Thanks for your hard work @yanshouwang! I'm considering using this plugin for an app I'm developing, but I want to know if the following scenario is supported or planned: As a central, connect to a peripheral that's already connected to the OS. I.e. I don't need to run discovery from the app, I just need to communicate with devices that are already connected.

I perused through the API and unless I missed something it doesn't seem like there's a way of doing it. Is there? Or do you have plans to implement it?

yanshouwang, 2024-04-10:
I think I can add a method like `retrieveConnectedPeripherals` in the next version.

yanshouwang, 2024-05-16:
The [6.0.0-dev.0](https://pub.dev/packages/bluetooth_low_energy/versions/6.0.0-dev.0) has released, the `retrieveConnectedPeripherals` is supported on all platforms except Windows.

yanshouwang, 2024-06-03:
Released in 6.0.0

closed 2024-06-03 by yanshouwang