yanshouwang/bluetooth_low_energy #62: 创建的Advertisement总是上一个

tilongzs, 2024-04-19:
例如在一个手机创建一个名字为“1”的广播：
```
final advertisement = Advertisement(
      name: '1',
    );
await PeripheralManager.instance.startAdvertising(advertisement);
```
几秒后停止广播：
```
await PeripheralManager.instance.stopAdvertising();
await PeripheralManager.instance.clearServices();
```
再创建一个名字为“2”的广播，另一个手机通过CentralManager.instance.discovered.listen()搜索到依然是"1“，直到停止广播，再创建一个名字为“3”的广播，另一个手机才搜索到”2“，而不是”3“。重启了也依然如此。

yanshouwang, 2024-04-19:
这是个已知问题，安卓上的广播名称来自蓝牙适配器的名称，我在开始广播前设置了蓝牙适配器的名称，但是不会立即生效，也没有任何方式监听适配器名称的改变，所以目前广播会使用上一次设置的名称（第一次会使用蓝牙设置里面的默认名称）

目前没有办法解决这个问题，如果有好的想法可以提出来

tilongzs, 2024-04-21:
谢谢，我看到了stackoverflow里的讨论。目前通过startAdvertising()后立即stopAdvertising()，再重新startAdvertising()就行了。

yanshouwang, 2024-06-03:
有时间可以测试一下 6.0.0 版本下这个问题是否还存在，新版本中名称改为存放在扫描回复中了

closed 2024-06-09 by yanshouwang