yanshouwang/bluetooth_low_energy #65: Crash on Windows When BLE Device Disconnects

dilip786, 2024-04-30:
On Windows noticed a crash when a BLE device disconnects unexpectedly.

**Issue Description:**

I've encountered a crash on Windows when a Bluetooth Low Energy (BLE) device disconnects unexpectedly. The crash does not always occur but happens randomly under the following conditions:

**Steps to Reproduce:**

- Connect to a BLE device.
- Perform several read and write operations.
- Disconnect the device.

[image]

yanshouwang, 2024-04-30:
This should be the same error as #63

dilip786, 2024-05-01:
Yes

yanshouwang, 2024-05-01:
> PlatformException(unknown, MyError, Stacktrace: ["0   bluetooth_low_energy_darwin         0x00000001069f46f4 $s27bluetooth_low_energy_darwin9wrapError33_67DD5082666B3ED7A98ED592AD08C18CLLySayypSgGypF + 1300", "1   bluetooth_low_energy_darwin         0x00000001069ff11c $s27bluetooth_low_energy_darwin28MyCentralManagerHostApiSetupC5setUp15binaryMessenger3apiySo013FlutterBinaryN0_p_AA0efghI0_pSgtFZyypSg_yAJctcfU10_ys6ResultOyyts5Error_pGcfU_ + 324", "2   bluetooth_low_energy_darwin         0x0000000106a13cf4 $ss6ResultOyyts5Error_pGIegg_ADytIegnr_TR + 40", "3   bluetooth_low_energy_darwin         0x0000000106a173b0 $ss6ResultOyyts5Error_pGytIegnr_ADIegg_TR + 84", "4   bluetooth_low_energy_darwin         0x0000000106a18b64 $s27bluetooth_low_energy_darwin16MyCentralManagerC23didDisconnectPeripheral7central10peripheral5errorySo09CBCentralG0C_So12CBPeripheralCs5Error_pSgtF + 5052", "5   bluetooth_low_energy_darwin         0x0000000106a22700 $s27bluetooth_low_energy_darwin24MyCentralManagerDelegateC07centralG0_23didDisconnectPeripheral5errorySo09CBCentralG0C_So12CBPeripheralCs5Error_pSgtF + 112", "6   bluetooth_low_energy_darwin         0x0000000106a227a8 $s27bluetooth_low_energy_darwin24MyCentralManagerDelegateC07centralG0_23didDisconnectPeripheral5errorySo09CBCentralG0C_So12CBPeripheralCs5Error_pSgtFTo + 144", "7   CoreBluetooth                       0x00000001a41c228c -[CBCentralManager handlePeripheralDisconnectionCompleted:] + 632", "8   CoreBluetooth                       0x00000001a41c40a0 -[CBCentralManager handleMsg:args:] + 244", "9   CoreBluetooth                       0x00000001a41b1aec -[CBManager xpcConnectionDidReceiveMsg:args:] + 192", "10  CoreBluetooth                       0x00000001a41b19e4 __30-[CBXpcConnection _handleMsg:]_block_invoke + 52", "11  libdispatch.dylib                   0x000000018f5e5cb8 _dispatch_call_block_and_release + 32", "12  libdispatch.dylib                   0x000000018f5e7910 _dispatch_client_callout + 20", "13  libdispatch.dylib                   0x000000018f5eeea4 _dispatch_lane_serial_drain + 748", "14  libdispatch.dylib                   0x000000018f5efa08 _dispatch_lane_invoke + 432", "15  libdispatch.dylib                   0x000000018f5f5eb8 _dispatch_main_queue_drain + 744", "16  libdispatch.dylib                   0x000000018f5f5bc0 _dispatch_main_queue_callback_4CF + 44", "17  CoreFoundation                      0x000000018f8b315c __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 16", "18  CoreFoundation                      0x000000018f870a80 __CFRunLoopRun + 1996", "19  CoreFoundation                      0x000000018f86fc5c CFRunLoopRunSpecific + 608", "20  HIToolbox                           0x0000000199dec448 RunCurrentEventLoopInMode + 292", "21  HIToolbox                           0x0000000199dec284 ReceiveNextEventCommon + 648", "22  HIToolbox                           0x0000000199debfdc _BlockUntilNextEventMatchingListInModeWithFilter + 76", "23  AppKit                              0x000000019304ac54 _DPSNextEvent + 660", "24  AppKit                              0x0000000193820ebc -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716", "25  AppKit                              0x000000019303e100 -[NSApplication run] + 476", "26  AppKit                              0x00000001930153cc NSApplicationMain + 880", "27  Aura 3D Imaging                     0x00000001046fd848 main + 12", "28  dyld                                0x000000018f4190e0 start + 2360"], null),#0      MyCentralManagerHostApi.writeCharacteristic (package:bluetooth_low_energy_darwin/src/my_api.g.dart:602:7)

This log is related to `bluetooth_low_energy_darwin` not `bluetooth_low_energy_windows`, is it related to this windows crash?

yanshouwang, 2024-05-16:
The [6.0.0-dev.0](https://pub.dev/packages/bluetooth_low_energy/versions/6.0.0-dev.0) has released, this issue should be resolved.

yanshouwang, 2024-06-03:
Fixed in 6.0.0

closed 2024-06-03 by yanshouwang