yanshouwang/bluetooth_low_energy #66: Error in Windows: platform channel messages

askandlearn, 2024-05-17:
Hello,
I found an issue using this package on Windows 10.
[√] Flutter (Channel stable, 3.16.9, on Microsoft Windows [Version 10.0.22631.3593], locale
    en-US)
[√] Windows Version (Installed version of Windows is version 10 or higher)
[√] Android toolchain - develop for Android devices (Android SDK version 33.0.0)
[√] Chrome - develop for the web
[√] Visual Studio - develop Windows apps (Visual Studio Community 2022 17.9.7)
[√] Android Studio (version 2021.2)
[√] VS Code (version 1.89.1)
[√] Connected device (4 available)
[√] Network resources

Steps to reproduce:
1) Run any flutter app which uses the CentralManager

Connecting to VM Service at ws://127.0.0.1:51793/-TN8a0R1R50=/ws
flutter: BLE: central manager state changed [ERROR:flutter/shell/common/shell.cc(1015)] The 'dev.flutter.pigeon.bluetooth_low_energy_windows.MyCentralManagerFlutterApi.onDiscovered' channel sent a message from native to Flutter on a non-platform thread. Platform channel messages must be sent on the platform thread. Failure to do so may result in data loss or crashes, and must be fixed in the plugin or application code creating that channel.
See https://docs.flutter.dev/platform-integration/platform-channels#channels-and-platform-threading for more information.
flutter: Data Directory: C:\Users\xxxx\OneDrive\Documents

yanshouwang, 2024-05-17:
This is blocked by this issue: https://github.com/flutter/flutter/issues/134346.

askandlearn, 2024-05-17:
Thanks for the reply @yanshouwang! 
The problem I have is that the bluetooth does not seem to work on Windows.  The Windows version is unable to discover my device.  If I compile the same app on Android, the device is detected.  This is the error message that was shown.  Were you able to discover devices on Windows? 

Thanks for all your work on this flutter package.  I have really liked it so far!

yanshouwang, 2024-05-17:
What's the version you are using? Can you provide the scanning codes?

yanshouwang, 2024-06-04:
Can you test this with 6.0.0 to see if this is resolved?

askandlearn, 2024-06-04:
Sure, I will try, but it may take some time because I'm a bit swamped.  The
new 6.0 pub version looks great! Appreciate all your efforts on this. Very
valuable library.