yanshouwang/bluetooth_low_energy #68: Support L2CAP

sampaioletti, 2024-05-22:
Would it be within the scope of this lib to support L2CAP communication? Looking into it for a prototype we are working on.

https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/logical-link-control-and-adaptation-protocol-specification.html

If it is, but not currently being looked at I'll try playing around with it and if successful I'll try and add a PR

yanshouwang, 2024-05-22:
I am not familiar with the L2CAP, it's great if you can implement this, just feel free to pull a new request. 
I know this is supported on Android and iOS. Is it possible to be implemented on Windows and Linux?

sampaioletti, 2024-05-22:
Technically I believe it's available on all platforms as it's just a couple layers down from GATT

But I'm just getting familiar with it as well. We have a protocol we are porting over from tcp and have 'mimicked' stream communication using standard GATT but appears that this will reduce some of the overhead/limitations of doing so.

[image]

pdalfarr, 2024-06-13:
Maybe this could help https://github.com/appsfactorygmbh/flutter-l2cap ?

yanshouwang, 2024-06-13:
> Maybe this could help https://github.com/appsfactorygmbh/flutter-l2cap ?

It's will be great is someone can contribute this feature to this plugin. I don't have much time to learn about the L2CAP at this moment.

I'm afraid this will be a long term goal if nobody want to help me with this, maybe v7.x or v8.x :sweat_smile:

pdalfarr, 2024-06-13:
Yep.

Keep focusing on the main goal of your project seems better indeed 😉