yanshouwang/bluetooth_low_energy #69: macos运行example报错

BraveAction, 2024-05-24:
### macos环境下报错:
lib/main.dart:459:12: Error: The method 'PopScope' isn't defined for the class '_PeripheralViewState'.
 - '_PeripheralViewState' is from 'package:bluetooth_low_energy_example/main.dart' ('lib/main.dart').
Try correcting the name to the name of an existing method, or defining a method named 'PopScope'.
    return PopScope(
           ^^^^^^^^
Target kernel_snapshot failed: Exception

Command PhaseScriptExecution failed with a nonzero exit code
warning: Run script build phase 'Run Script' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase. (in target 'Flutter Assemble' from project 'Runner')
** BUILD FAILED **

Building macOS application...
Exception: Build process failed

### 这是我的flutter版本:
Flutter 3.13.6 • channel stable • https://github.com/flutter/flutter.git
Framework • revision ead455963c (8 months ago) • 2023-09-26 18:28:17 -0700
Engine • revision a794cf2681
Tools • Dart 3.1.3 • DevTools 2.25.0

yanshouwang, 2024-05-24:
The PopScope widget is released after 3.16.0.

There are two suggestions for you

1. Just remove the PopScope widget or use WillPopScope instead.
2. Upgrade flutter version.

I will use go_router instead of this widget in the next versoin.

yanshouwang, 2024-06-03:
Fixed in 6.0.0

closed 2024-06-03 by yanshouwang