yanshouwang/bluetooth_low_energy #70: 6.0.0-dev.1 build error

Shunt22, 2024-06-02:
Hi!
I was looking for a way to connect without scanning, by creating my own Peripheral with known UUID (I know you can't do this on IOS etc), I need this for Windows.

It looks likes on 6.0.0 version I can do it with Peripheral class so I switch to 6.0.0 version and got this error when building on Windows:

```
Pub/Cache/hosted/pub.dev/bluetooth_low_energy_android-6.0.0-dev.4/lib/src/my_api.dart(90,33): error GC2F972A8: The argument type 'ManufacturerSpecificData?' can't be assigned to the parameter type 'List<ManufacturerSpecificData>'. 
Pub/Cache/hosted/pub.dev/bluetooth_low_energy_android-6.0.0-dev.4/lib/src/my_api.dart(227,68): error GE5CFE876: The method 'toArgs' isn't defined for the class 'List<ManufacturerSpecificData>
Pub/Cache/hosted/pub.dev/bluetooth_low_energy_darwin-6.0.0-dev.2/lib/src/my_api.dart(65,33): error GC2F972A8: The argument type 'ManufacturerSpecificData?' can't be assigned to the parameter type 'List<ManufacturerSpecificData>'
Pub/Cache/hosted/pub.dev/bluetooth_low_energy_darwin-6.0.0-dev.2/lib/src/my_api.dart(225,68): error GE5CFE876: The method 'toArgs' isn't defined for the class 'List<ManufacturerSpecificData>'.
Pub/Cache/hosted/pub.dev/bluetooth_low_energy_linux-6.0.0-dev.2/lib/src/my_bluez.dart(64,33): error GC2F972A8: The argument type 'ManufacturerSpecificData?' can't be assigned to the parameter type 'List<ManufacturerSpecificData>'. 
```

yanshouwang, 2024-06-02:
Now this plugin doesn't support create peripheral without scan. Create peripheral without scan is not normal way in BLE, I know we can create BluetoothDevice with MAC address on Android, but this method still scan for the device, so why not just scan and filter the device with the address(UUID)? Also the MAC address maybe a random address and change every few minutes, it's not a good idea to create peripheral with MAC address.

There is a breaking change in the latest `bluetooth_low_enregy_interface` now, you need add the last version to make the plugin to work.

``` Dart
bluetooth_low_energy_platform_interface: 6.0.0-dev.18
```

I will fix this issue soon.

Shunt22, 2024-06-03:
Indeed, I can scan first and filter by UUID.  I've done it this way.
Will wait for 6.0.0 stable release.
Thanks for the package!

closed 2024-06-03 by Shunt22