yanshouwang/bluetooth_low_energy #71: iOS and Android peripheral manager exposes other platform services

ekuleshov, 2024-06-02:
I'm using BLE peripheral manager to expose a single service and advertise peripheral under my own custom name in Android and iOS apps. Also making sure to clear existing services before starting BLE advertise.

However when I connect to that BLE peripheral, I see multiple services, including battery service, device info, current time, some audio stuff, etc.

That is triggering some unwanted side effects, such as my peripheral is recognized as an audio device by the car audio, and other things.

This is also reproducible with the `bluetooth_low_energy` own example app when peripheral is started.

[image]

yanshouwang, 2024-06-02:
These services are added by system. I don't think there is a way to remove them.

ekuleshov, 2024-06-24:
> These services are added by system. I don't think there is a way to remove them.
> 
> See the discussion in this [doc](https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/removeallservices()?language=objc)

I was hoping there is a way to have a separate instance of peripheral manager independent from the one used by the system or other apps.

yanshouwang, 2024-06-24:
> I was hoping there is a way to have a separate instance of peripheral manager independent from the one used by the system or other apps.
> 
> E.g. multi-advertising on Android. Re: [BluetoothAdapter#isMultipleAdvertisementSupported()](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#isMultipleAdvertisementSupported()), AdvertisingSet, etc.
> 
> https://source.android.com/docs/core/connect/bluetooth/ble_advertising

I don't think there is a way to do that. The advertisement has no effect with GATT services, we can't modify system services through multi-advertising.