yanshouwang/bluetooth_low_energy #72: Windows requirements

Shunt22, 2024-06-03:
I have two PCs, both windows 10.
Using bluetooth_low_energy: ^5.0.7
On home PC  - I am using your package in my application on Windows and it works fine.
On work PC (Windows 10 19042.789), when I try to build my application on Windows - I got this error:

```
my_central_manager.obj : error LNK2019: ��뫪� �� ��ࠧ�襭�� ���譨� ᨬ��� "__declspec(dllimport) public: __cdecl std::bad_cast::bad_cast(char const *)" (__imp_??0bad_cast@std@@QEAA@PEBD@Z) � �㭪樨 "private: e
num winrt::Windows::Devices::Bluetooth::GenericAttributeProfile::GattWriteOption __cdecl bluetooth_low_energy_windows::MyCentralManager::m_write_type_number_args_to_write_option(__int64)" (?m_write_type_number_args_to_write_option@MyCentralManager@bluetooth_low_energy_windows@@AEAA?AW4GattWriteOption@GenericAttributeProfile@Bluetooth@Devices@Windows@winrt@@_J@Z). 
```

What's wrong?

yanshouwang, 2024-06-03:
LNK2019 means the compiler can't resolve the GattWriteOption API, but the 19042.789 should support this API.

Is there any other useful logs? or run flutter build windows -v to see more details about this error.

Also it's better to set the system encoding with UTF-8(beta) in the locale settings to avoid those unicodes.

Shunt22, 2024-06-03:
I've updated Visual Studio build tools and this issuse is gone.
But there is another one:
```
[+10392 ms] ...\windows\flutter\ephemeral\.plugin_symlinks\bluetooth_low_energy_windows\windows\my_central_manager.cpp(678,33): error C2039: "format": �� ���� 童���"std". ...\build\windows\x64\plugins\bluetooth_low_energy_windows\bluetooth_low_energy_windows_plugin.vcxproj]
[ +139 ms] C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.31.31103\include\variant(30): message : �. ������� "std"
...\build\windows\x64\plugins\bluetooth_low_energy_windows\bluetooth_low_energy_windows_plugin.vcxproj]
[  +61 ms] ...\windows\flutter\ephemeral\.plugin_symlinks\bluetooth_low_energy_windows\windows\my_central_manager.cpp(678,39): error C3861: format: �����䨪��� ��
������ ...
\build\windows\x64\plugins\bluetooth_low_energy_windows\bluetooth_low_energy_windows_plugin.vcxproj]
[ +297 ms] ...\windows\flutter\ephemeral\.plugin_symlinks\bluetooth_low_energy_windows\windows\my_central_manager.cpp(687,33): error C2039: "format": �� ���� 童���
"std". ...\build\windows\x64\plugins\bluetooth_low_energy_windows\bluetooth_low_energy_windows_plugin.vcxproj]
[ +430 ms] C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.31.31103\include\variant(30): message : �. ������� "std"
```

A lot of  error C3861: format: and error C2039: "format": errors.

Flutter doctor:
```
[√] Flutter (Channel stable, 3.22.1, on Microsoft Windows [Version 10.0.19042.789], locale ru-RU)
    • Flutter version 3.22.1 on channel stable at D:\Home\Andrew\FlutterSDK
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision a14f74ff3a (12 days ago), 2024-05-22 11:08:21 -0500
    • Engine revision 55eae6864b
    • Dart version 3.4.1
    • DevTools version 2.34.3

[√] Windows Version (Installed version of Windows is version 10 or higher)

[√] Android toolchain - develop for Android devices (Android SDK version 35.0.0-rc4)
    • Android SDK at D:\Home\Andrew\AndroidSDK
    • Platform android-34, build-tools 35.0.0-rc4
    • Java binary at: D:\Programs\AndroidStudio2023\jbr\bin\java
    • Java version OpenJDK Runtime Environment (build 17.0.10+0--11572160)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files (x86)\Google\Chrome\Application\chrome.exe

[√] Visual Studio - develop Windows apps (Visual Studio Build Tools 2022 17.3.5)
    • Visual Studio at C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools
    • Visual Studio Build Tools 2022 version 17.3.32922.545
    • Windows 10 SDK version 10.0.22621.0

[√] Android Studio (version 2023.3)
    • Android Studio at D:\Programs\AndroidStudio2023
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 17.0.10+0--11572160)

[√] IntelliJ IDEA Community Edition (version 2020.2)
    • IntelliJ at D:\Programs\IntelliJ IDEA Community Edition 2020.2.3
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart

[√] Connected device (3 available)
    • Windows (desktop) • windows • windows-x64    • Microsoft Windows [Version 10.0.19042.789]
    • Chrome (web)      • chrome  • web-javascript • Google Chrome 125.0.6422.113
    • Edge (web)        • edge    • web-javascript • Microsoft Edge 124.0.2478.105

[√] Network resources
    • All expected network resources are available.

• No issues found!
```

yanshouwang, 2024-06-03:
There is a related issue #73

yanshouwang, 2024-06-04:
Is this issue still exists with 6.0.0? The #73 has resolved with the latest version

Shunt22, 2024-06-04:
@yanshouwang 
After updating to 6.0.0 working on both PCs.
Thanks!

closed 2024-06-04 by Shunt22