yanshouwang/bluetooth_low_energy #76: 為何要用ViewModel?這樣改跟5.x差異太大

spiderman16888, 2024-06-05:
比如說下面的程式碼.整個改變太大.
本來可以不用   ViewModel 嗎？
  GoRoute(
                  path: ':uuid',
                  builder: (context, state) {
                    final uuidValue = state.pathParameters['uuid']!;
                    final uuid = UUID.fromString(uuidValue);
                    final viewModel =
                        ViewModel.of<CentralManagerViewModel>(context);
                    final eventArgs = viewModel.discoveries.firstWhere(
                        (discovery) => discovery.peripheral.uuid == uuid);
                    return ViewModelBinding(
                      viewBuilder: (context) => PeripheralView(),
                      viewModelBuilder: (context) =>
                          PeripheralViewModel(eventArgs),
                    );
                  },

yanshouwang, 2024-06-05:
只是在 example 代码里使用了 MVVM 架构，实际使用中并不需要使用 MVVM，只需要根据迁移文档修改代码即可

spiderman16888, 2024-06-05:
好的，我試試.
但我找遷移文件.
https://pub.dev/documentation/bluetooth_low_energy/latest/doc/migrations/migration-v6.md
他顯示404 Not Found.

spiderman16888, 2024-06-05:
我找到了.我試試看遷移.謝謝.
https://github.com/yanshouwang/bluetooth_low_energy/blob/master/bluetooth_low_energy/doc/migrations/migration-v6.md

closed 2024-06-05 by spiderman16888