yanshouwang/bluetooth_low_energy #82: Bug when calling removeAllServices()

afekhar, 2024-06-23:
Bug when calling PeripheralManager.removeAllServices()

`Exception has occurred.
PlatformException (PlatformException(IllegalStateException, java.lang.IllegalStateException, Cause: null, Stacktrace: java.lang.IllegalStateException
	at dev.hebei.bluetooth_low_energy_android.MyPeripheralManager.getServer(MyPeripheralManager.kt:82)
	at dev.hebei.bluetooth_low_energy_android.MyPeripheralManager.removeAllServices(MyPeripheralManager.kt:183)
	at dev.hebei.bluetooth_low_energy_android.MyPeripheralManagerHostAPI$Companion.setUp$lambda-20$lambda-19(MyAPI.g.kt:1351)
	at dev.hebei.bluetooth_low_energy_android.MyPeripheralManagerHostAPI$Companion.$r8$lambda$0UBEBMYjxvY0FDQwOvliV4DN4bs(Unknown Source:0)
	at dev.hebei.bluetooth_low_energy_android.MyPeripheralManagerHostAPI$Companion$$ExternalSyntheticLambda4.onMessage(Unknown Source:2)
	at io.flutter.plugin.common.BasicMessageChannel$IncomingMessageHandler.onMessage(BasicMessageChannel.java:261)
	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:292)
	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:319)
	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
	at android.os.Handler.handleCallback(Handler.java:958)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:294)
	at android.app.ActivityThread.main(ActivityThread.java:8177)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
, null))`

yanshouwang, 2024-06-23:
Is the bluetooth state is `poweredOn` when you call the `removeAllServices()`? this error will throw if the bluetooth is poweredOff or the `BluetoothManager.openGattServer` returns null,

Can you provide the log info details when this issue occured?

Seems this is the same issue as #41

afekhar, 2024-06-23:
Bluetooth state was in an authorized state. Thank you very much for the hint. I close the issue.

closed 2024-06-23 by afekhar