yanshouwang/bluetooth_low_energy #83: PeripheralManager.startAdvertising throws NullPointerException on Android

ekuleshov, 2024-06-24:
I've got a `NullPointerException` reported from some user's devices (Galaxy Tab A 7.0 SM-T280) when `PeripheralManager.startAdvertising` was called.

According to Android docs the [BluetoothAdapter.getBluetoothLeAdvertiser()](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBluetoothLeAdvertiser()) can return null if Bluetooth is turned off or if Bluetooth LE Advertising is not supported on this device.

It would be nice if unsupported advertising handled more gracefully.

```
PlatformException: PlatformException(NullPointerException, 
  java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeAdvertiser.startAdvertising(android.bluetooth.le.AdvertiseSettings, android.bluetooth.le.AdvertiseData, android.bluetooth.le.AdvertiseData, android.bluetooth.le.AdvertiseCallback)' 
  on a null object reference, Cause: null, 
  Stacktrace: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeAdvertiser.startAdvertising(android.bluetooth.le.AdvertiseSettings, android.bluetooth.le.AdvertiseData, android.bluetooth.le.AdvertiseData, android.bluetooth.le.AdvertiseCallback)' 
    on a null object reference
	at d5.z1.z(MyPeripheralManager.kt:41)
	at d5.z2$a.x(MyAPI.g.kt:52)
	at d5.z2$a.i(MyAPI.g.kt)
	at d5.q2.a(R8$$SyntheticClass)
	at r5.a$b.a(BasicMessageChannel.java:18)
	at f5.c.l(DartMessenger.java:19)
	at f5.c.m(DartMessenger.java:42)
	at f5.c.i(DartMessenger.java)
	at f5.b.run(R8$$SyntheticClass)
	at android.os.Handler.handleCallback(Handler.java:739)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:145)
	at android.app.ActivityThread.main(ActivityThread.java:6939)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1404)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1199)
, null)
  File "my_api.g.dart", line 1548, in MyPeripheralManagerHostAPI.startAdvertising
  File "<asynchronous suspension>"
  File "my_peripheral_manager.dart", line 176, in MyPeripheralManager.startAdvertising
  File "<asynchronous suspension>"
  ...
```

yanshouwang, 2024-06-24:
Thanks for this report, I didn't notice that PeripheralManager is unsupported on some devices, it's supported on all the devices I have tested. I will add this check in the next version, but I don't have a way to test this.

ekuleshov, 2024-06-24:
Thank you. Unfortunately I don't have that device either (running Android 4.1.x). The report came from the crash reporting system. I wonder if the `startAdvertising` call should return some kind of status to indicate if advertising is supported or not.

closed 2024-06-30 by yanshouwang

ekuleshov, 2024-07-14:
It looks like I'm still getting this error with `6.0.2` version. This one came from Xiaomi 22101316G running Android 14.

```
PlatformException: PlatformException(NullPointerException, java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeAdvertiser.startAdvertising(android.bluetooth.le.AdvertiseSettings, android.bluetooth.le.AdvertiseData, android.bluetooth.le.AdvertiseData, android.bluetooth.le.AdvertiseCallback)' on a null object reference, Cause: null, Stacktrace: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeAdvertiser.startAdvertising(android.bluetooth.le.AdvertiseSettings, android.bluetooth.le.AdvertiseData, android.bluetooth.le.AdvertiseData, android.bluetooth.le.AdvertiseCallback)' on a null object reference
	at x2.z1.o(MyPeripheralManager.kt:41)
	at x2.z2$a.x(MyAPI.g.kt:52)
	at x2.z2$a.i(Unknown Source:0)
	at x2.q2.a(Unknown Source:2)
	at l3.a$b.a(BasicMessageChannel.java:18)
	at z2.c.l(DartMessenger.java:19)
	at z2.c.m(DartMessenger.java:42)
	at z2.c.i(Unknown Source:0)
	at z2.b.run(Unknown Source:12)
	at android.os.Handler.handleCallback(Handler.java:958)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:222)
	at android.os.Looper.loop(Looper.java:314)
	at android.app.ActivityThread.main(ActivityThread.java:8706)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:565)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1081)
, null)
  File "my_api.g.dart", line 1548, in MyPeripheralManagerHostAPI.startAdvertising
  File "<asynchronous suspension>"
  File "my_peripheral_manager.dart", line 176, in MyPeripheralManager.startAdvertising
  File "<asynchronous suspension>"
```
```

ekuleshov, 2024-07-14:
And another variant from Amazon tablet running Android 9.

```
PlatformException: PlatformException(IllegalStateException, java.lang.IllegalStateException, Cause: null, Stacktrace: java.lang.IllegalStateException
	at x2.z1.R(MyPeripheralManager.kt:8)
	at x2.z1.i(MyPeripheralManager.kt:1)
	at x2.z2$a.w(MyAPI.g.kt:6)
	at x2.z2$a.g(Unknown Source:0)
	at x2.p2.a(Unknown Source:2)
	at l3.a$b.a(BasicMessageChannel.java:18)
	at z2.c.l(DartMessenger.java:19)
	at z2.c.m(DartMessenger.java:42)
	at z2.c.i(Unknown Source:0)
	at z2.b.run(Unknown Source:12)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:211)
	at android.app.ActivityThread.main(ActivityThread.java:6775)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
, null)
  File "my_api.g.dart", line 1526, in MyPeripheralManagerHostAPI.removeAllServices
  File "<asynchronous suspension>"
  File "my_peripheral_manager.dart", line 151, in MyPeripheralManager.removeAllServices
  File "<asynchronous suspension>"
```

yanshouwang, 2024-07-15:
Have you checked the bluetooth state before advertising?

commit 78e74f77 (2024-06-30): Use `isMultipleAdvertisementSupported` to check whether `PeripheralManager` is supported on this device, closes #83 (#85)

* Use `isMultipleAdvertisementSupported` to check whether `PeripheralManager` is supported on this device.

* 6.0.2

* 6.0.2

---------