yanshouwang/bluetooth_low_energy #84: StartDiscovery throws IllegalStateException on Android in example app

itslenny, 2024-06-26:
Hi, I'm just getting started with this library. To try it out I cloned this repository and launched the example app.

On Android the "Central" mode throws an IllegalStateException when I click "begin"

Clicking begin on "Peripheral" mode seems to work as expected (or at least doesn't throw an error)

I also ran it on on MacOS everything seems to run as expected there.

I'm going to try digging into this more on my end, but hoped maybe you'd have some more insight.

The following is the stack dump. Above is some logs I added in just to see the state through authorization and start discovery.

```
The Flutter DevTools debugger and profiler on Pixel 6a

I/flutter ( 5018): @@@@@@ _stateChangedSubscription
I/flutter ( 5018): ... state = unauthorized , idx = 2, event = unauthorized

I/flutter ( 5018): ....AUTHORIZING!!!!
D/CompatibilityChangeReporter( 5018): Compat change id reported: 78294732; UID 10377; state: ENABLED
I/flutter ( 5018): ....AUTHORIZING!!!!..... complete. result=true

I/flutter ( 5018): @@@@@@ _stateChangedSubscription
I/flutter ( 5018): ... state = poweredOn , idx = 4, event = poweredOn

I/flutter ( 5018): @@@@@@ startDiscovery ... 1 (start)
I/flutter ( 5018): ... state = poweredOn , idx = 4
I/flutter ( 5018): @@@@@@ startDiscovery ... 2 (b4)

D/BluetoothAdapter( 5018): isLeEnabled(): ON
D/BluetoothLeScanner( 5018): onScannerRegistered() - status=128 scannerId=0 mScannerId=0

E/flutter ( 5018): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(IllegalStateException, java.lang.IllegalStateException: Start discovery failed with error code: 2, Cause: null, Stacktrace: java.lang.IllegalStateException: Start discovery failed with error code: 2
E/flutter ( 5018): 	at dev.hebei.bluetooth_low_energy_android.MyCentralManager.onScanFailed(MyCentralManager.kt:360)
E/flutter ( 5018): 	at dev.hebei.bluetooth_low_energy_android.MyScanCallback.onScanFailed(MyScanCallback.kt:15)
E/flutter ( 5018): 	at android.bluetooth.le.BluetoothLeScanner$1.run(BluetoothLeScanner.java:645)
E/flutter ( 5018): 	at android.os.Handler.handleCallback(Handler.java:959)
E/flutter ( 5018): 	at android.os.Handler.dispatchMessage(Handler.java:100)
E/flutter ( 5018): 	at android.os.Looper.loopOnce(Looper.java:232)
E/flutter ( 5018): 	at android.os.Looper.loop(Looper.java:317)
E/flutter ( 5018): 	at android.app.ActivityThread.main(ActivityThread.java:8592)
E/flutter ( 5018): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter ( 5018): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
E/flutter ( 5018): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:878)
E/flutter ( 5018): , null)
E/flutter ( 5018): #0      MyCentralManagerHostAPI.startDiscovery (package:bluetooth_low_energy_android/src/my_api.g.dart:726:7)
E/flutter ( 5018): <asynchronous suspension>
E/flutter ( 5018): #1      MyCentralManager.startDiscovery (package:bluetooth_low_energy_android/src/my_central_manager.dart:97:5)
E/flutter ( 5018): <asynchronous suspension>
E/flutter ( 5018): #2      CentralManagerViewModel.startDiscovery (package:bluetooth_low_energy_example/view_models/central_manager_view_model.dart:66:5)
E/flutter ( 5018): <asynchronous suspension>
E/flutter ( 5018): #3      CentralManagerView.build.<anonymous closure> (package:bluetooth_low_energy_example/views/central_manager_view.dart:32:23)
E/flutter ( 5018): <asynchronous suspension>
E/flutter ( 5018):
```

yanshouwang, 2024-06-26:
Here is the [docs about scan failed with status 2](https://developer.android.com/reference/android/bluetooth/le/ScanCallback#SCAN_FAILED_APPLICATION_REGISTRATION_FAILED).

You can try to reopen the BLE or restart the phone to see if this can be resolved.

itslenny, 2024-06-27:
Thanks for the info @yanshouwang 

Since I had a consistent repro I tried everything I could to fix it programmatically to see if maybe I could provide a fix for this state, but nothing worked.

In the end the only way to fix this issue (for me) was to restart the phone, and then everything worked fine. Seems to be a bug with Android, and as far as I can tell the only real option is to tell the user to restart their phone if they get in this state.

closed 2024-06-27 by itslenny