yanshouwang/bluetooth_low_energy #86: IllegalStateException on Android

ekuleshov, 2024-07-14:
The following `IllegalStateException` on some customer devices reported in Sentry. Can't reproduce this on my own devices. Using `bluetooth_low_energy` 6.0.2.

It looks like the `mBluetoothGattServerCallback` may not be set at the time it is called:

https://github.com/yanshouwang/bluetooth_low_energy/blob/78e74f7793312394e6221de89560cd7c9e5791cf/bluetooth_low_energy_android/android/src/main/kotlin/dev/hebei/bluetooth_low_energy_android/MyPeripheralManager.kt#L27-L35

Dart exception:
```
File "my_api.g.dart", line 1482, in MyPeripheralManagerHostAPI.addService
  File "<asynchronous suspension>"
  File "my_peripheral_manager.dart", line 135, in MyPeripheralManager.addService
  File "<asynchronous suspension>"
```

Android exception:

```
PlatformException(IllegalStateException, java.lang.IllegalStateException, Cause: null, Stacktrace: java.lang.IllegalStateException
	at x2.z1.z(MyPeripheralManager.kt:30)
	at x2.z2$a.u(MyAPI.g.kt:30)
	at x2.z2$a.b(Unknown Source:0)
	at x2.n2.a(Unknown Source:2)
	at l3.a$b.a(BasicMessageChannel.java:18)
	at z2.c.l(DartMessenger.java:19)
	at z2.c.m(DartMessenger.java:42)
	at z2.c.i(Unknown Source:0)
	at z2.b.run(Unknown Source:12)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
```

yanshouwang, 2024-07-15:
I think this error is throwed by line 82 or line 165, is the bluetooth is poweredOn when before you call the `addService` method?

https://github.com/yanshouwang/bluetooth_low_energy/blob/78e74f7793312394e6221de89560cd7c9e5791cf/bluetooth_low_energy_android/android/src/main/kotlin/dev/hebei/bluetooth_low_energy_android/MyPeripheralManager.kt#L82

https://github.com/yanshouwang/bluetooth_low_energy/blob/78e74f7793312394e6221de89560cd7c9e5791cf/bluetooth_low_energy_android/android/src/main/kotlin/dev/hebei/bluetooth_low_energy_android/MyPeripheralManager.kt#L160-L171

yanshouwang, 2024-07-15:
> ```
>     PeripheralManager pm = PeripheralManager();
> 
>     pm.stateChanged.listen((event) async {
>       final state = event.state;
>       if (Platform.isAndroid && state == BluetoothLowEnergyState.unauthorized) {
>         await pm.authorize();
>       }
>       if (event.state == BluetoothLowEnergyState.poweredOn) {
>         _initBlePeripheral(pm);
>       } else {
>         _disposeBlePeripheral(pm);
>       }
>     });
> 
>     BluetoothLowEnergyState state = pm.state;
>     if (state == BluetoothLowEnergyState.poweredOn) {
>       await _initBlePeripheral(pm);
>     }
> ```

The code seems right, so this error was throwed by the line 165(addService returns false)?

Need more information about this or how to reproduce this issue with some steps

yanshouwang, 2024-07-15:
Yes, I think I should add some description to each exception. Also maybe you can upload the debug symbols to Sentry to get the entire stack trace?

See this [docs](https://docs.sentry.io/platforms/flutter/upload-debug/) for details

ekuleshov, 2024-07-15:
> upload the debug symbols to Sentry to get the entire stack trace?

Already did that. Will have to wait for a few days when users will pickup a new app release...

ekuleshov, 2024-07-28:
Here is the most recent unobfuscated stacktrace with bluetooth_low_energy: 6.0.2

```
PlatformException: PlatformException(IllegalStateException, java.lang.IllegalStateException, Cause: null, Stacktrace: java.lang.IllegalStateException
	at dev.hebei.bluetooth_low_energy_android.MyPeripheralManager.getServer(MyPeripheralManager.java:82)
	at dev.hebei.bluetooth_low_energy_android.MyPeripheralManager.removeAllServices(MyPeripheralManager.java:185)
	at dev.hebei.bluetooth_low_energy_android.MyPeripheralManagerHostAPI$Companion.setUp$lambda$20$lambda$19(MyPeripheralManagerHostAPI.java:1351)
	at dev.hebei.bluetooth_low_energy_android.MyPeripheralManagerHostAPI$Companion.$r8$lambda$OnoTl9oJ8yjgRSaVQDt0sDSzAFs(Unknown Source:0)
	at dev.hebei.bluetooth_low_energy_android.MyPeripheralManagerHostAPI$Companion$$InternalSyntheticLambda$1$7f73d5bfd3ddafa853578382aa442d6d585ec1fa0e35f9033c4f2c22db4dbd19$9.onMessage(Unknown Source:2)
	at io.flutter.plugin.common.BasicMessageChannel$IncomingMessageHandler.onMessage(BasicMessageChannel.java:261)
	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:292)
	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0(DartMessenger.java:319)
	at io.flutter.embedding.engine.dart.DartMessenger.$r8$lambda$2j2MERcK825A5j1fv5sZ7xB2Iuo(Unknown Source:0)
	at io.flutter.embedding.engine.dart.DartMessenger$$InternalSyntheticLambda$1$5dd8b6f7959f08bc8717eff7469e77e06ef5aed51cc0cee17f1e13794798223f$0.run(Unknown Source:12)
...
```

```
  #0      MyPeripheralManagerHostAPI.removeAllServices (package:bluetooth_low_energy_android/src/my_api.g.dart:1526)
  File "<asynchronous suspension>"
  #2      MyPeripheralManager.removeAllServices (package:bluetooth_low_energy_android/src/my_peripheral_manager.dart:151)
  File "<asynchronous suspension>"
...
```

yanshouwang, 2024-07-29:
> @yanshouwang I think so, here is my init code below. The `removeAllServices` and `addService` are inside `_initBlePeripheral()`
> 
> ```
>     PeripheralManager pm = PeripheralManager();
> 
>     pm.stateChanged.listen((event) async {
>       final state = event.state;
>       if (Platform.isAndroid && state == BluetoothLowEnergyState.unauthorized) {
>         await pm.authorize();
>       }
>       if (event.state == BluetoothLowEnergyState.poweredOn) {
>         _initBlePeripheral(pm);
>       } else {
>         _disposeBlePeripheral(pm);
>       }
>     });
> 
>     BluetoothLowEnergyState state = pm.state;
>     if (state == BluetoothLowEnergyState.poweredOn) {
>       await _initBlePeripheral(pm);
>     }
> ```

This code seems has an issue when the state is unauthorized. After the authorize mehtod complete, the app will run into the else block which will call the `_disposeBlePeripheral` method.

If the user denied the permission, you will run into this error.

ekuleshov, 2024-07-29:
> This code seems has an issue when the state is unauthorized. After the authorize mehtod complete, the app will run into the else block which will call the `_disposeBlePeripheral` method.

Are you suggesting that `PeripheralManager.stopAdvertising()` and `PeripheralManager.removeAllServices()` methods require BLE permission be granted on Android?

yanshouwang, 2024-07-29:
> Are you suggesting that `PeripheralManager.stopAdvertising()` and `PeripheralManager.removeAllServices()` methods require BLE permission be granted on Android?

All methods require permission when being called, the `if else` blocks in your code should not be called when state is unauthorized. You can use `switch` instead of `if else`, or use `else if` instead of the second `if`

ekuleshov, 2024-07-29:
> All methods require permission when being called, the `if else` blocks in your code should not be called when state is unauthorized. You can use `switch` instead of `if else`, or use `else if` instead of the second `if`

It is counter-intuitive to require permission on destructing/removing methods...

E.g. when the app BLE permission is manually removed in the system settings and then BLE power goes off - user would be asked for BLE permission in order to shut things down? It would be more natural to not request permission in that case.

yanshouwang, 2024-07-29:
There is an issue in your code. You call authorize when status is unauthorized, and you should return after that. But you still check the state after that. The event.state is still unauthorized.

If the state is still denied by user. The state will keep unauthorized. Then you need show a warning UI to let user jump to the system settings to grant the permission.

If the permission is granted by user. You will receive another state event with poweredOn or poweredOff. If the state is poweredOff. You should show a warning UI to let user turn on the Bluetooth.

You should only continue the business when state is poweredOn.

yanshouwang, 2024-07-29:
> E.g. when the app BLE permission is manually removed in the system settings and then BLE power goes off - user would be asked for BLE permission in order to shut things down? It would be more natural to not request permission in that case.

You don't need to dispose anything when BLE is turned off. Just show a warning message to user is OK.

yanshouwang, 2024-08-07:
You need to request permission when the state is unauthorized. If the permission is authorized by user, the state will be poweredOn or poweredOff. Call removeAllServices when the state is poweredOn. Do not call any methods when state is poweredOff.

ekuleshov, 2024-08-07:
> You need to request permission when the state is unauthorized. If the permission is authorized by user, the state will be poweredOn or poweredOff. Call removeAllServices when the state is poweredOn. Do not call any methods when state is poweredOff.

It look odd... When BLE state is powered off I probably should not be asking user's permission for bluetooth?

How does the stage transition look like when the app does not have BLE permissions and BLE us powered on/off?  Could you please add the complete state chart to the plugin documentation.

Also, after `unauthorized` state is raised and user grants permission, does framework automatically trigger the `poweredOn` state?

Having an `unauthorized` along with other power and availability states is somewhat cumbersome to manage. It would have been simpler and more intuitive to manage if `unauthorized` won't be a state, but some kind of separate property/stream.

yanshouwang, 2024-08-07:
> It look odd... When BLE state is powered off I probably should not be asking user's permission for bluetooth?
>
> How does the stage transition look like when the app does not have BLE permissions and [...]

This behavior is the same as the [CoreBlutooth](https://developer.apple.com/documentation/corebluetooth/cbmanagerstate). The state will always unauthorized if the user doesn't agree the app to use the BLE. After authorized, the state will update to poweredOff or poweredOn automatically.

ekuleshov, 2024-08-07:
> This behavior is the same as the [CoreBlutooth](https://developer.apple.com/documentation/corebluetooth/cbmanagerstate). The state will always be unauthorized if the user doesn't agree the app to use the BLE. After authorized, the state will update to poweredOff or poweredOn automatically.
> 
> See this [discussion](https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/centralmanagerdidupdatestate(_:)#Discussion) for more details.

Hmm. That discussion is related to iOS, yet the `unauthorized` state is Android only?...

I'm still unclear if the state change event will be send after user gives permission after an `unauthorized` event is processed and `PeripheralManager.authorize()` is called?

```mermaid
graph TD;
  unauthorized --> poweredOff
  unauthorized --> poweredOn
  poweredOff --> poweredOn
  poweredOff --> unauthorized
  poweredOn --> poweredOff
  poweredOn --> unauthorized
  unknown
  unsupported
```

yanshouwang, 2024-08-08:
I think the graph should like this.
The app will be killed by system when the permission is disallowed by user from the app settings.

``` mermaid
graph TD;
unknown --> unsupported & unauthorized & poweredOff & poweredOn
unauthorized --> poweredOff & poweredOn
poweredOff <--> poweredOn
```

ekuleshov, 2024-08-08:
> The app will be killed by system when the permission is disallowed by user from the app settings.

That sounds scary... In other BLE libs removing Bluetooth permission simply cut off the scan/connectivity but I don't see why the app should be killed.

Also, under what circumstances the application will get the `unknown` state event?

yanshouwang, 2024-08-08:
> That sounds scary... In other BLE libs removing Bluetooth permission simply cut off the scan/connectivity but I don't see why the app should be killed.

The system will restart the app when user remove any permission of the app. This is a normal behavior on Android and iOS
> 
> 
> Also, under what circumstances the application will get the `unknown` state event?

The unknown state is the default value. The state changed event will never trigger this value.

yanshouwang, 2024-08-08:
> > That sounds scary... In other BLE libs removing Bluetooth permission simply cut off the scan/connectivity but I don't see why the app should be killed.
> 
> The system will restart the app when user remove any permission of the app. This is a normal behavior on Android and iOS. It's unrelated to plugins.
> > 
> > 
> > Also, under what circumstances the application will get the `unknown` state event?
> 
> The unknown state is the default value. The state changed event will never trigger this value.
>

ekuleshov, 2024-09-08:
With the latest `bluetooth_low_energy` and the following init code I'm still getting reports on `IllegalStateException` happening in the app running on user's Android devices.

```
    PeripheralManager pm = PeripheralManager();
    pm.stateChanged.listen((event) async => _updateBlePeripheral(event.state, pm));
    await _updateBlePeripheral(pm.state, pm);
    ...
    
  Future<void> _updateBlePeripheral(BluetoothLowEnergyState state, PeripheralManager pm) async {
    if (state == BluetoothLowEnergyState.unauthorized) {
      if (System.isAndroid) {
        await pm.authorize();
      }
    } else if (state == BluetoothLowEnergyState.poweredOn) {
      await _initBlePeripheral(pm);
    } else {
      await _disposeBlePeripheral(pm);
    }
  }

  Future<void> _initBlePeripheral(PeripheralManager pm) async {
     ...
     await pm.addService(GATTService(...));  // <--- failing here
     ...
  }    
```

Here is the exception stacktace

```
PlatformException: PlatformException(IllegalStateException, java.lang.IllegalStateException,
   Cause: null, Stacktrace: java.lang.IllegalStateException
	at v3.z1.e(MyPeripheralManager.kt:30)
	at v3.z2$a.u(MyAPI.g.kt:30)
	at v3.z2$a.b(Unknown Source:0)
	at v3.n2.a(Unknown Source:2)
	at j4.a$b.a(BasicMessageChannel.java:18)
	at x3.c.l(DartMessenger.java:19)
	at x3.c.m(DartMessenger.java:42)
	at x3.c.i(Unknown Source:0)
	at x3.b.run(Unknown Source:12)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7870)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1009)
, null)

  #0      MyPeripheralManagerHostAPI.addService (package:bluetooth_low_energy_android/src/my_api.g.dart:1482)
  File "<asynchronous suspension>"
  #2      MyPeripheralManager.addService (package:bluetooth_low_energy_android/src/my_peripheral_manager.dart:135)
  File "<asynchronous suspension>"
  #4      ...._initBlePeripheral (...)
  File "<asynchronous suspension>"
```

yanshouwang, 2024-09-09:
> ```
> PlatformException: PlatformException(IllegalStateException, java.lang.IllegalStateException,
>    Cause: null, Stacktrace: java.lang.IllegalStateException
> 	at v3.z1.e(MyPeripheralManager.kt:30) [...]

I can't know what exactly caused this error from the stacktrace. It's not easy to fix this without a procedure to reproduce this.

ekuleshov, 2024-09-09:
> I can't know what exactly caused this error from the stacktrace. It's not easy to fix this without a procedure to reproduce this.

This is literally happening on the app startup. I wasn't been able to reproduce it locally and only getting the  Sentry crash reports from user's devices. A Google Pixel 6 Pro phone running android 12 in this case.

ekuleshov, 2025-04-05:
I'm still getting IllegalStateException on Android. 
From the Android stacktrace it looks like the `mServer` value is null, but the adapter state on the Dart side is `BluetoothLowEnergyState.poweredOn`

https://github.com/yanshouwang/bluetooth_low_energy/blob/a7700eb9152eb98bce1ba2d090e7a20e4b6da787/bluetooth_low_energy_android/android/src/main/kotlin/dev/hebei/bluetooth_low_energy_android/MyPeripheralManager.kt#L82

```
MyPeripheralManagerHostAPI.removeAllServices (my_api.g.dart:1526)
MyPeripheralManager.removeAllServices (my_peripheral_manager.dart:151)
_initBlePeripheral ()
_updateBlePeripheral ()
```

```
java.lang.IllegalStateException
at dev.hebei.bluetooth_low_energy_android.MyPeripheralManager.getServer(MyPeripheralManager.java:82)
at dev.hebei.bluetooth_low_energy_android.MyPeripheralManager.removeAllServices(MyPeripheralManager.java:185)
at dev.hebei.bluetooth_low_energy_android.MyPeripheralManagerHostAPI$Companion.setUp$lambda$20$lambda$19(MyPeripheralManagerHostAPI.java:1351)
at B3.Pigeon$PdfxApi$$ExternalSyntheticLambda5.onMessage(Pigeon.java:0)
at io.flutter.plugin.common.BasicMessageChannel$IncomingMessageHandler.onMessage(BasicMessageChannel.java:261)
at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:292)
at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0(DartMessenger.java:319)
at android.os.Handler.handleCallback(Handler.java:969)
```

I can't tell if there were any errors in the `_api.openGATTServer()` call, but if there were some errors, it seems like the state on the Dart side is kept as `poweredOn` and exception is logged and then ignored:

https://github.com/yanshouwang/bluetooth_low_energy/blob/a7700eb9152eb98bce1ba2d090e7a20e4b6da787/bluetooth_low_energy_android/lib/src/my_peripheral_manager.dart#L969-L976

Here is the code that trigger this error:

```
   // initialization
    pm.stateChanged.listen((event) async => _updateBlePeripheral(event.state));
    unawaited(_updateBlePeripheral(pm.state));
```

```
  Future<void> _updateBlePeripheral(BluetoothLowEnergyState state) async {
    if (state == BluetoothLowEnergyState.unauthorized) {
      if (System.isAndroid) {
        PeripheralManager pm = PeripheralManager();
        bool res = await pm.authorize();
        debugPrint('## BLE authorized $res');
      }
    } else if (state == BluetoothLowEnergyState.poweredOn) {
      await _initBlePeripheral();
    }
  }
```

```
  Future<void> _initBlePeripheral() async {
    PeripheralManager pm = PeripheralManager();

    try {
      await pm.removeAllServices();
    } on PlatformException catch (error, stackTrace) {
      ...
    }
```

arjunmnath, 2025-08-13:
@yanshouwang  any updates on the issue?