yanshouwang/bluetooth_low_energy #87: Prevent unauthorized connections

CrayolaEater, 2024-07-16:
Is there a way to prevent unauthorized connection? Or just prevent a certain incoming connection request?

yanshouwang, 2024-07-16:
Currently we use a handshake in the business logic or heart break to keep the connection as a trick.

I don't know is there a way to do this with BLE itself.

Levi-Lesches, 2024-08-29:
@yanshouwang you may want to reconsider the stale bot as it closes issues that don't have updates, even if they're valid requests that might be worked on...

yanshouwang, 2024-08-29:
> @yanshouwang you may want to reconsider the stale bot as it closes issues that don't have updates, even if they're valid requests that might be worked on...

I'll disable this behavior later. Thanks for your suggestion.