yanshouwang/bluetooth_low_energy #88: CentralManager.connectionStateChanged not called when connection to peripheral is lost

CrayolaEater, 2024-07-18:
I have a subscription on CentralManager.connectionStateChanged. It is called when state is connected, however is not called when the state is disconnected.

yanshouwang, 2024-07-18:
The disconnected event is not called immediately when device disconnected, it should received a few seconds later.

Can you test this with the example and provide more information about this like app logs to trace this issue.

yanshouwang, 2025-07-29:
> [@yanshouwang](https://github.com/yanshouwang) Is this relying on supervisionTimeout, or is there explicit logic when disconnect is emitted?

The connectionStateChanged event is triggered by the OS, It's never occured in my project.

taukakao, 2025-07-31:
After debugging this for a while it seems that connectionStateChanged works as intended, but it can look like it doesn't.

In my case, I want to connect two Android devices. If I close the app on one device, then Android will just remove the service, not close the connection. So that means that I have to check if the service still exists every time I want to write to my characteristic. This behavior does actually make sense if you think about it but might be counter intuitive.

It would be nice though if you could monitor changes in the GATT Services of a peripheral with this library. I haven't found a way to do so.

yanshouwang, 2025-07-31:
As @taukakao said, if you close the app on the peripheral side, the OS maybe not disconnect the central device but just remove the service from GATT database, in this case, the CentralManager will not be able to listen state changed event, let me know if this issue is caused by this situation.