yanshouwang/bluetooth_low_energy #89: StartAdvertising doesn't have a status callback for success or failure.

15972078884, 2024-07-19:
Can it be added?

yanshouwang, 2024-07-19:
The `startAdvertising` method returns a Future<void>, this future will complete when the status is success or throws an error when the status is failed. You can try catch this method and await it to know the status.

danbachar, 2025-11-19:
> Do you mean the startAdvertising future is never completed or you can't find the advertisement on the central side?

The `startAdvertising` future is never completed - it just hangs there.

yanshouwang, 2025-11-20:
> > Do you mean the startAdvertising future is never completed or you can't find the advertisement on the central side?
> 
> The `startAdvertising` future is never completed - it just hangs there.

I'll test this later, I think maybe this is caused by the thread model change in recent flutter versions

danbachar, 2025-11-20:
Upon further debugging I found out this problem arises when the user requests both peripheral and central authorization. Isn't this supported by the library?