yanshouwang/bluetooth_low_energy #95: Android connection priority

mpecar, 2024-08-23:
I am using your plugin for some time now and it is working very well!

I see that you are working on version 7, but one thing that I really need on Android is still missing. Can you please expose this:
https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestConnectionPriority(int)

The story behind this is similar to "why do we ever need to set the MTU?". I am not sure what happens, but my guess would be: I have peripheral devices that use high priority, but they somehow fail to notify the central. The result is that central misses data packets. I can fix this by setting the priority to CONNECTION_PRIORITY_HIGH, but not with your package.

Any chance of including this in v7?

yanshouwang, 2024-08-23:
No problem, I'll add this method in the next version.

mpecar, 2024-08-24:
Great!

srmncnk, 2024-11-18:
@yanshouwang Any updates or ETAs on this?

yanshouwang, 2024-11-19:
> @yanshouwang Any updates or ETAs on this?

I'm working on other projects these days. The next version still need some days.

closed 2025-03-25 by yanshouwang