yanshouwang/bluetooth_low_energy #96: retrieveConnectedPeripherals for Windows

wojciechwaliszek-digica, 2024-08-28:
Hi! 
Thank you for your great work. I would like to find out if you plan to support the function `retrieveConnectedPeripherals` on Windows devices. I could really use it. Thanks.

yanshouwang, 2024-08-28:
@wojciechwaliszek-digica  I can't find an [winrt API](https://learn.microsoft.com/zh-cn/uwp/api/windows.devices.bluetooth?view=winrt-26100) to implement the `retrieveConnectedPeripherals` method. Let me know If you have any idea about how to impelment this on Windows.

wojciechwaliszek-digica, 2024-08-29:
I have never worked with winrt api. I did some research using LLM models. here's what I have got:
- You can use the DeviceInformation.FindAllAsync method to get a list of paired Bluetooth devices. This will give you information about devices that are paired with the system, and some of them might be connected.
- You can subscribe to the DeviceWatcher or BluetoothLEAdvertisementWatcher to monitor when devices connect or disconnect in real time. This way, you can keep track of connected peripherals.

I don't have any experience here, so I don't know if this will be useful to you. I hope it helps.

yanshouwang, 2024-08-29:
> I have never worked with winrt api. I did some research using LLM models. here's what I have got:
>
> - You can use the DeviceInformation.FindAllAsync method to get a list of paired Bluetooth devices. [...]

I can try the first way to see if it's possible to get the device connection state. 

The second way is impossible to important this method.

wojciechwaliszek-digica, 2024-09-09:
Hi! Have you got any updates on this? Did the method work?

yanshouwang, 2024-09-11:
> * DeviceInformation.FindAllAsync

I have looked at the API document today, I think it can be implementd with `DeviceInformation.FindAllAsync`. But it will take some time, since I'm working on other jobs these days.

wojciechwaliszek-digica, 2024-09-11:
Sounds great! Thanks!

yanshouwang, 2024-09-13:
I have implemented this in the newest dev branch.

**_Notice this method will take a very long time on Windows. It not a retrieve action but a find all action actually, the system will find devices continuously about 30s in my test._**

```
final stopwatch = Stopwatch()..start();
    final peripherals = await _manager.retrieveConnectedPeripherals();
    stopwatch.stop();
    logger.info(
        'retrieveConnectedPeripherals length ${peripherals.length}, takes ${stopwatch.elapsed}');
    for (var peripheral in peripherals) {
      logger.info('retrieveConnectedPeripherals ${peripheral.uuid}');
    }
```

[image]

closed 2024-09-13 by yanshouwang

wojciechwaliszek-digica, 2024-09-26:
I tried to run my project using the 7.0.0 branch but I'm getting a lot of errors connected to logger, or some missing definitions:
```
/C:/Users/user/AppData/Local/Pub/Cache/hosted/pub.dev/bluetooth_low_energy_windows-7.0.0-dev.0/lib/bluetooth_low_energy_windows.dart(8,5): error GA0A3EB9C: Undefined name 'PlatformCentralManager'. [C:\application\build\windows\x64\flutter\flutter_assemble.vcxproj]
/C:/Users/user/AppData/Local/Pub/Cache/hosted/pub.dev/bluetooth_low_energy_windows-7.0.0-dev.0/lib/bluetooth_low_energy_windows.dart(9,5): error GA0A3EB9C: Undefined name 'PlatformPeripheralManager'. [C:\application\build\windows\x64\flutter\flutter_assemble.vcxproj]
...
/C:/Users/uesr/AppData/Local/Pub/Cache/hosted/pub.dev/bluetooth_low_energy_android-7.0.0-dev.0/lib/src/my_central_manager.dart(419,7): error GD65BB2B6: The getter 'logger' isn't defined for the class 'MyCentralManager'. [C:\application\build\windows\x64\flutter\flutter_assemble.vcxproj]
/C:/Users/user/AppData/Local/Pub/Cache/hosted/pub.dev/bluetooth_low_energy_android-7.0.0-dev.0/lib/src/my_central_manager.dart(423,7): error GD65BB2B6: The getter 'logger' isn't defined for the class 'MyCentralManager'. [C:\application\build\windows\x64\flutter\flutter_assemble.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): error MSB8066: Custom build for 'C:\application\build\windows\x64\CMakeFiles\65f587699271c26d78d8724f36ad3571\flutter_windows.dll.rule;C:\application\build\windows\x64\CMakeFiles\45b7e51f32d334c96ad36e286c1e116f\flutter_assemble.rule' exited with code 1. [C:\application\build\windows\x64\flutter\flutter_assemble.vcxproj]
Error: Build process failed.
```

yanshouwang, 2024-09-26:
@wojciechwaliszek-digica The branch is under development. I'll fix the issues as soon as possible.