yanshouwang/bluetooth_low_energy #98: Cant send write requests to multiple devices on Android/IOS

IvanTorres21, 2024-08-30:
Hello,

I was able to connect two different bluetooth devices to the same app and send write requests to them in the windows version. But when executing the same application within the Android/iOS version an error pops out when trying to write to the first bluetooth device I connected to. 

Steps to reproduce:
1.- Connect to one device
2.- Connect to a second device
3.- Send a write request to both
4.- The second device recives the requests, while the first one throws this error:

`flutter: PlatformException(illegalArgument, MyError, Stacktrace: ["0   bluetooth_low_energy_darwin         0x0000000103cf3250 $s27bluetooth_low_energy_darwin9wrapError33_D3FF39C2E9589F2A07964E0F3A1AEE9ELLySayypSgGypF + 1708", "1   bluetooth_low_energy_darwin         0x0000000103d029fc $s27bluetooth_low_energy_darwin28MyCentralManagerHostAPISetupC5setUp15binaryMessenger3api20messageChannelSuffixySo013FlutterBinaryM0_p_AA0efgH3API_pSgSStFZyypSg_yAKctcfU14_ys6ResultOyyts5Error_pGcfU_ + 324", "2   bluetooth_low_energy_darwin         0x0000000103d1fa1c $s27bluetooth_low_energy_darwin16MyCentralManagerC19writeCharacteristic8uuidArgs08hashCodeK005valueK004typeK010completionySS_s5Int64VSo24FlutterStandardTypedDataCAA0e27GATTCharacteristicWriteTypeK0Oys6ResultOyyts5Error_pGctF + 1104", "3   bluetooth_low_energy_darwin         0x0000000103d2acb4 $s27bluetooth_low_energy_darwin16MyCentralManagerCAA0efG7HostAPIA2aDP19writeCharacteristic8uuidArgs08hashCodeM005valueM004typeM010completionySS_s5Int64VSo24FlutterStandardTypedDataCAA0e27GATTCharacteristicWriteTypeM0Oys6ResultOyyts5Error_pGctFTW + 32", "4   bluetooth_low_energy_darwin         0x0000000103d02888 $s27bluetooth_low_energy_darwin28MyCentralManagerHostAPISetupC5setUp15binaryMessenger3api20messageChannelSuffixySo013FlutterBinaryM0_p_AA0efgH3API_pSgSStFZyypSg_yAKctcfU14_ + 1760", "5   bluetooth_low_energy_darwin         0x0000000103cfe278 $sypSgAAIegn_Iegng_yXlSgABIeyBy_IeyByy_TR + 236", "6   Flutter                             0x00000001072f84c8 __48-[FlutterBasicMessageChannel setMessageHandler:]_block_invoke + 160", "7   Flutter                             0x0000000106d65794 ___ZN7flutter25PlatformMessageHandlerIos21HandlePlatformMessageENSt3_fl10unique_ptrINS_15PlatformMessageENS1_14default_deleteIS3_EEEE_block_invoke + 116", "8   libdispatch.dylib                   0x00000001c606e7a8 B51E7CDB-ABC9-35AF-B8BB-2DCE23BC4D6E + 407464", "9   libdispatch.dylib                   0x00000001c606f780 B51E7CDB-ABC9-35AF-B8BB-2DCE23BC4D6E + 411520", "10  libdispatch.dylib                   0x00000001c601bbdc B51E7CDB-ABC9-35AF-B8BB-2DCE23BC4D6E + 68572", "11  libdispatch.dylib                   0x00000001c601b82c B51E7CDB-ABC9-35AF-B8BB-2DCE23BC4D6E + 67628", "12  CoreFoundation                      0x00000001bf3519ac 55B9BA28-4C5C-3FE7-9C47-4983337D6E83 + 596396", "13  CoreFoundation                      0x00000001bf335648 55B9BA28-4C5C-3FE7-9C47-4983337D6E83 + 480840", "14  CoreFoundation                      0x00000001bf339d20 CFRunLoopRunSpecific + 584", "15  GraphicsServices                    0x00000001f6b49998 GSEventRunModal + 160", "16  UIKitCore                           0x00000001c15cc448 27A9C298-B702-3C39-8C06-07196E4CD16B + 3609672", "17  UIKitCore                           0x00000001c15cc0c0 UIApplicationMain + 312", "18  Runner                              0x0000000102c7895c main + 64", "19  dyld                                0x00000001dcae4344 8A4B89B7-D348-375B-97B1-FC8A84E3E5CE + 82756"], null)
flutter: 
#0      MyCentralManagerHostAPI.writeCharacteristic (package:bluetooth_low_energy_darwin/src/my_api.g.dart:861:7)
<asynchronous suspension>
#1      MyCentralManager.writeCharacteristic (package:bluetooth_low_energy_darwin/src/my_central_manager.dart:183:5)
<asynchronous suspension>`

This is my pubsec.yaml.

```
name: TestApp
description: "A new Flutter project."
publish_to: 'none' # Remove this line if you wish to publish to pub.dev

version: 2.6.16+124823265

environment:
  sdk: '>=3.4.0 <4.0.0'

dependencies:
  flutter:
    sdk: flutter

  cupertino_icons: ^1.0.2
  change_app_package_name: ^1.1.0
  firebase_core: ^2.25.5
  firebase_auth: ^4.17.6
  firebase_database: ^10.4.7
  bluetooth_low_energy: ^6.0.0
  udp: ^5.0.3
  package_info_plus: ^8.0.0
  http: ^1.2.0
  intl: ^0.19.0
  binary: ^3.0.1
  shared_preferences: ^2.2.2
  provider: ^6.1.2
  firebase_storage: ^11.6.9
  fast_cached_network_image: ^1.2.9
  path_provider: ^2.1.2
  shimmer: ^3.0.0
  firebase_crashlytics: ^3.4.18
  firebase_analytics: ^10.8.9
  webview_windows: ^0.4.0
  window_manager: ^0.3.8
  webview_flutter: ^4.7.0
  firebase_remote_config: ^4.3.17
  responsive_sizer: ^3.3.1
  flutter_svg: ^2.0.10+1
  animations: ^2.0.11
  upgrader: ^10.3.0
  wakelock_plus: ^1.2.5
  json_annotation: ^4.9.0
  device_info_plus: ^10.1.0
  jiffy: ^6.3.1
  url_launcher: ^6.2.6
  configcat_client: ^4.1.1
dev_dependencies:
  flutter_test:
    sdk: flutter
  integration_test:
[... 1199 lines cut ...]
      url: "https://pub.dev"
    source: hosted
    version: "1.0.4"
  xml:
    dependency: transitive
    description:
      name: xml
      sha256: b015a8ad1c488f66851d762d3090a21c600e479dc75e68328c52774040cf9226
      url: "https://pub.dev"
    source: hosted
    version: "6.5.0"
  yaml:
    dependency: transitive
    description:
      name: yaml
      sha256: "75769501ea3489fca56601ff33454fe45507ea3bfb014161abc3b43ae25989d5"
      url: "https://pub.dev"
    source: hosted
    version: "3.1.2"
sdks:
  dart: ">=3.4.0 <4.0.0"
  flutter: ">=3.22.0"
`

```

yanshouwang, 2024-08-30:
Can you provide a sample to reproduce this issue. From the logs above, it seems you passed invalid Peripheral or GATTCharacteristic instance, so the plugin can not find corresponding native instance which will throw illegalArgument error.

IvanTorres21, 2024-09-02:
Here is the issue replicated on a single page. 

```dart
import 'dart:io';
import 'dart:typed_data';

import 'package:bluetooth_low_energy/bluetooth_low_energy.dart';
import 'package:flutter/material.dart';

class ReplicateIssue extends StatefulWidget {
  @override
  State<ReplicateIssue> createState() => _ReplicateIssueState();
}

class _ReplicateIssueState extends State<ReplicateIssue> {
  final List<DiscoveredEventArgs> _devices = List.empty(growable: true);
  final List<DiscoveredEventArgs> _connectedTo = List.empty(growable: true);
  List<GATTService> services = List.empty();
  final String commonCharacteristic = ""; // This is the common characteristic, both of the devices have this.
  final String commonCharacteristic2 = ""; // Same as above, they share this characteristic.

  bool isScanning = false;

  @override
  void initState() {
    scanDevices();
    super.initState();
  }

  @override
  Widget build(BuildContext context) {
    return Scaffold(
      body: Column(
        children: [
          ElevatedButton(onPressed: sendMessage, child: Text("Send BLE Message")),
          SizedBox(
            height: MediaQuery.of(context).size.height * .6,
            child: ListView.builder(
              itemBuilder: (_, index) =>
                  ElevatedButton(onPressed: () => connectDevice(_devices[index]), child: Text("${_devices[index].advertisement.name}")),
              itemCount: _devices.length,
            ),
          )
        ],
      ),
    );
  }

  bool isWritting = false;
  Future<void> sendMessage() async {
    if(isWritting) return;
    isWritting = true;
    for(var device in _connectedTo) {
      await Bluetooth.writeCharacteristic(device.peripheral, getGattCharacteristic(commonCharacteristic2),  value: Uint8List.fromList([0x0]), type: GATTCharacteristicWriteType.withResponse);
    }
    isWritting = false;
  }

  void scanDevices() async {
    if(!Platform.isWindows) {
      await Bluetooth.authorize();
    }
    isScanning = true;
    Bluetooth.discovered.listen(AddDevice);
    while(isScanning) {
      await Future.delayed(const Duration(seconds: 1));
      await Bluetooth.startDiscovery();
      await Future.delayed(const Duration(seconds: 2));
      await Bluetooth.stopDiscovery();
    }
  }

   CentralManager get Bluetooth => CentralManager();

  void AddDevice(DiscoveredEventArgs event) {
    print(event.advertisement.name);
    if (_devices.any((x) => x.advertisement.name == event.advertisement.name)) return;
    setState(() {
      _devices.add(event);
    });
  }

  Future<void> connectDevice(DiscoveredEventArgs device) async {
    isScanning = false;
    await Bluetooth.stopDiscovery();

    await Bluetooth.connect(device.peripheral);
    services = await Bluetooth.discoverGATT(device.peripheral); //Both the devices we are connecting to are the same.
    _connectedTo.add(device);
    await Bluetooth.writeCharacteristic(device.peripheral, getGattCharacteristic(commonCharacteristic),  value: Uint8List.fromList([0x25]), type: GATTCharacteristicWriteType.withResponse);
  }

  GATTCharacteristic getGattCharacteristic(String whichOne) {
    for (var service in services) {
      for (var characteristic in service.characteristics) {
        if (UUID(characteristic) == whichOne) return characteristic;
      }
    }
    throw new Exception("Characteristic not found in any service");
  }

  String UUID(GATTCharacteristic gatt) => gatt.uuid.toString().toUpperCase();
}
```

As I said previously, when executing this in windows it works perfectly fine. But mobile throws the error. I´m using the same bluetooth devices for both attempts, and the bluetooth devices are the exact same.

yanshouwang, 2024-09-02:
There are issues in these two method.

> ```dart
> Future<void> connectDevice(DiscoveredEventArgs device) async {
>     isScanning = false;
>     await Bluetooth.stopDiscovery();
> 
>     await Bluetooth.connect(device.peripheral);
>     services = await Bluetooth.discoverGATT(device.peripheral); //Both the devices we are connecting to are the same.
>     _connectedTo.add(device);
>     await Bluetooth.writeCharacteristic(device.peripheral, getGattCharacteristic(commonCharacteristic),  value: Uint8List.fromList([0x25]), type: GATTCharacteristicWriteType.withResponse);
>   }
> 
>   GATTCharacteristic getGattCharacteristic(String whichOne) {
>     for (var service in services) {
>       for (var characteristic in service.characteristics) {
>         if (UUID(characteristic) == whichOne) return characteristic;
>       }
>     }
>     throw new Exception("Characteristic not found in any service");
>   }
> ```

1. The `services` is set after the first deice connected, and will be replaced after the second device connected. As a result, the first device's service instances is lost.
2. Since the two devices have the same services defination. When you get service for the first device, you can get a service with UUID from the `services` without error, but the service is belong to the second's actually. So the plugin will throw this error when you tell it to write with the first device and the service which belong to the second device.

This issus is not exist on Windows because the way I used to find corresponding native instance on Windows is different from the way on other platforms.
1. On Windows, I use device address to find  device instances, use GATT attribute handle to find GATT attribute instances. That means if the tow devices have the same service, characteristic and descriptor definations, the attribute handles are same too. So we can find the service of the first device with the second device's service without error in this case.
2. On other platforms, I use object's hashCode to find the corresponding native instance. That's will cause the service can't be found with the second devie's service, so this error occured.
3. On windows, it's not easy to use the hashCode because the c++ object doesn't have this. On other platforms, it's not possible to use the GATT attribute handle because the Bluetooth APIs doesn't have this.

IvanTorres21, 2024-09-03:
Ahhh, I see. Sorry then, I did the windows implementation first and since it worked thought it was a problem with the plugin itself. 

Sorry for the issue, I´ll store the services twice then.

closed 2024-09-03 by IvanTorres21