yanshouwang/bluetooth_low_energy #104: 安卓平板无法使用ble外围设备模式

runoob-coder, 2024-10-21:
我用您的demo在安卓平板上运行了下，主机模式扫描连接都没有问题，但是外围模式（从机）无法使用，显示说是不支持

[image]

但是我用nRF Connect看了下平板的设备信息，是支持外围模式（从机）的

[image]

请问这个问题怎么解决呢？目前的需求是将安卓平板设为外围模式并广播，但是此包无法使用

yanshouwang, 2024-10-21:
6.0.2 版本增加了对 `isMultipleAdvertisementSupport` 的判断，根据官方文档的描述，需要检查此属性来判断设备是否支持 LE 广播，参考 https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBluetoothLeAdvertiser()

从你的截图来看这个设备不支持 Multiple Advertisement, 因此会显示不支持状态。

你可以尝试切换到 6.0.1 分支来测试一下。

runoob-coder, 2024-10-21:
> 6.0.2 版本增加了对 `isMultipleAdvertisementSupport` 的判断，根据官方文档的描述，需要检查此属性来判断设备是否支持 LE 广播，参考 https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBluetoothLeAdvertiser()
> 
> 从你的截图来看这个设备不支持 Multiple Advertisement, 因此会显示不支持状态。
> 
> 你可以尝试切换到 6.0.1 分支来测试一下。

我切换到bluetooth_low_energy_android: ^6.0.1，可以作为外围模式并被其他手机设备扫描到了，我再研究研究下

谢谢大佬🙏