yanshouwang/bluetooth_low_energy #106: Error calling the startAdvertising() method

Yolo-cell-hash, 2024-10-28:
I'm trying to advertise my device data to ble devices but can't seem to get it done, any help is appreciated ...

```

  late final BluetoothLowEnergyManager bluetoothDevice ;

  final advertisement = Advertisement(
    name: Platform.isAndroid ? 'Device Name' : null,
    serviceData: {
      // mServiceUUID : Uint8List.fromList([0x82,0x45,0x65,0x87,0x45,0x65,0x87,0x45,0x65,0x87,0x45,0x65,0x87,0x45,0x65,0x87]),
    },
    // serviceUUIDs: [mServiceUUID],
    manufacturerSpecificData: Platform.isIOS || Platform.isMacOS ? [] : [
      ManufacturerSpecificData(id: 0x2e09, data: Uint8List.fromList([0x01,0x24,0x78,0x87,]))
    ]

  );

void startAdvertising() async{
    await bluetoothDevice.startAdvertising(advertisement);
  }

```

I'm getting the following error -

`The method 'startAdvertising' isn't defined for the type 'BluetoothLowEnergyManager'. (Documentation)`

yanshouwang, 2024-10-28:
Which platform are you running?

Yolo-cell-hash, 2024-10-28:
I'm running it on Android , I've updated the comment here,   ` late final PeripheralManager bluetoothDevice ;` and now the code builds but I cant find my device on nrF or so..?

Yolo-cell-hash, 2024-10-28:
E/flutter (24720): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: LateInitializationError: Field 'bluetoothDevice' has not been initialized.
E/flutter (24720): #0      AdvertiseData.bluetoothDevice (package:flutter_furniture_locks/utils/advertise_data.dart)
E/flutter (24720): #1      AdvertiseData.startAdvertising (package:flutter_furniture_locks/utils/advertise_data.dart:31:11)
E/flutter (24720): #2      _HomeScreenState.build.<anonymous closure> (package:flutter_furniture_locks/screens/home_screen.dart:51:29)
E/flutter (24720): #3      _InkResponseState.handleTap (package:flutter/src/material/ink_well.dart:1170:21)
E/flutter (24720): #4      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:351:24)
E/flutter (24720): #5      TapGestureRecognizer.handleTapUp (package:flutter/src/gestures/tap.dart:656:11)
E/flutter (24720): #6      BaseTapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:313:5)
E/flutter (24720): #7      BaseTapGestureRecognizer.handlePrimaryPointer (package:flutter/src/gestures/tap.dart:246:7)
E/flutter (24720): #8      PrimaryPointerGestureRecognizer.handleEvent (package:flutter/src/gestures/recognizer.dart:703:9)
E/flutter (24720): #9      PointerRouter._dispatch (package:flutter/src/gestures/pointer_router.dart:98:12)
E/flutter (24720): #10     PointerRouter._dispatchEventToRoutes.<anonymous closure> (package:flutter/src/gestures/pointer_router.dart:143:9)
E/flutter (24720): #11     _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:633:13)
E/flutter (24720): #12     PointerRouter._dispatchEventToRoutes (package:flutter/src/gestures/pointer_router.dart:141:18)
E/flutter (24720): #13     PointerRouter.route (package:flutter/src/gestures/pointer_router.dart:127:7)
E/flutter (24720): #14     GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:501:19)
E/flutter (24720): #15     GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:481:22)
E/flutter (24720): #16     RendererBinding.dispatchEvent (package:flutter/src/rendering/binding.dart:450:11)
E/flutter (24720): #17     GestureBinding._handlePointerEventImmediately (package:flutter/src/gestures/binding.dart:426:7)
E/flutter (24720): #18     GestureBinding.handlePointerEvent (package:flutter/src/gestures/binding.dart:389:5)
E/flutter (24720): #19     GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:336:7)
E/flutter (24720): #20     GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:305:9)
E/flutter (24720): #21     _invoke1 (dart:ui/hooks.dart:328:13)
E/flutter (24720): #22     PlatformDispatcher._dispatchPointerDataPacket (dart:ui/platform_dispatcher.dart:442:7)
E/flutter (24720): #23     _dispatchPointerDataPacket (dart:ui/hooks.dart:262:31)
D/[FBP-Android](24720): [FBP] onMethodCall: flutterRestart
D/[FBP-Android](24720): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](24720): [FBP] connectedPeripherals: 0

closed 2024-10-28 by Yolo-cell-hash