yanshouwang/bluetooth_low_energy #109: startDiscovery fails on android

vincentvulog, 2024-11-12:
In the android sample application, calling    await _manager.startDiscovery trigger the following exception (and scanning does not work)

Using Android 14 (5 sept 2024)
BLE on
Rights granted at application startup

PlatformException (PlatformException(IllegalStateException, java.lang.IllegalStateException: Start discovery failed with error code: 2, Cause: null, Stacktrace: java.lang.IllegalStateException: Start discovery failed with error code: 2
	at dev.hebei.bluetooth_low_energy_android.MyCentralManager.onScanFailed(MyCentralManager.kt:360)
	at dev.hebei.bluetooth_low_energy_android.MyScanCallback.onScanFailed(MyScanCallback.kt:15)
	at android.bluetooth.le.BluetoothLeScanner$1.run(BluetoothLeScanner.java:650)
	at android.os.Handler.handleCallback(Handler.java:958)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:257)
	at android.os.Looper.loop(Looper.java:368)
	at android.app.ActivityThread.main(ActivityThread.java:8839)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:572)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1049)
, null))

vincentvulog, 2024-11-12:
[image]

yanshouwang, 2024-11-14:
Duplicated of #84

closed 2024-11-14 by vincentvulog