yanshouwang/bluetooth_low_energy #112: Reading notification value from peripheral

HyeonJungHam, 2024-12-06:
I want to read the characteristic notified value from peripheral, 
but it seems only the central manager includes the notified value in the subscription. 
Can you tell me how I can read the notified value? Is there any workaround?

yanshouwang, 2024-12-06:
What do you mean by read the notified value?

The notified values should be listened through the characteristicNotified stream.

Or you can call the readCharacteristic or writeCharacteristic to read/write peripherals.

HyeonJungHam, 2024-12-06:
Oops, sorry, I need to explain more about what I want to do. I want to implement a peripheral which advertises to communicate with an external central device. I want to communicate with notification. What I want to ask is that if I want to use the peripheral side only, is there a way to listen to the notified value?

I found that notified values could be read from the central manager, but not from the peripheral manager. It seems the peripheral manager only can listen the notify state (not value) through  characteristicNotifyStateChanged of Stream<GATTCharacteristicNotifyStateChangedEventArgs> type.

yanshouwang, 2024-12-06:
> I found that notified values could be read from the central manager, but not from the peripheral manager. It seems the peripheral manager only can listen the notify state (not value) through characteristicNotifyStateChanged of Stream type.

Yes, The peripheral is responsible to notify subscribed centrals that it's value has changed. And the central side can change the notify state and subscribe the notify stream.

What is you should do is listen the central's write event. You can listen the PeripheralManager.characteristicWriteRequested event.

You can see the [PeripheralViewModel](https://github.com/yanshouwang/bluetooth_low_energy/blob/master/bluetooth_low_energy/example/lib/view_models/peripheral_manager_view_model.dart) in the example to get more details about how to listen centrals write requests and how to respone them.

HyeonJungHam, 2024-12-10:
Yes, I realized I don't have to notify from both sides. It works well, and I like your coding style. Very alike with the actual Kotlin bindings, thank you!

closed 2024-12-10 by HyeonJungHam