yanshouwang/bluetooth_low_energy #116: Cannot achieve automated connection and write

RobinSteiger, 2024-12-19:
Hi,

I'm trying to write a Flutter bloc to achieve automatic connection, write and confirmation between to mobile, for both Android and iOS platform. 
I was able to test successfully the respond of my PeripheralManager to the correct write, but I can't achieve to establish a connection with the CentralManager in order to send a write request. I suspect part of the problem is the change of role, but I cannot block the peripheral changing with my semaphore because connectionStateChanged is not available for iOS. I also receive a "trying to send a message from platform API to a non-API platform" when I'm testing the connection between a Windows instance running the CentralManager and an Android instance instance running the PeripheralManager.
I've set static uuid to find the right characteristics but I'm still unable to achieve the correct behavior. 

Here's my connection function : 

Future<void> _onAddFriendDetection(
    AddFriendDetection event, 
    Emitter<AddFriendState> emit
    ) async {

      emit(state.copyWith(status: AddFriendStatus.detection, displayMessage: 'Searching ...'));

      // Changing role permission (semaphore)
      bool changingRole = true;
      // Define role, true = central, false = peripheral
      bool role = true;
      // Max turn 
      int maxTurn = 3;

      // Central Stream Subscription
      final StreamSubscription centralSubscription = _centralManager.discoveriesStream.listen((eventArgs) async {
        final peripheral = eventArgs.peripheral;
        final advertisement = eventArgs.advertisement;
        final name = advertisement.name!;
        print("(Central) Detected name : $name");
        if (name == state.friendName) {
          changingRole = false;
          emit(state.copyWith(status : AddFriendStatus.found));
          await _centralManager.stopDiscovery();
          final connect = await _centralManager.connect(peripheral);
          final canWrite = await _centralManager.discoverGATT();
          if (connect && canWrite) {
            final writeResult = await _centralManager.writeRequest(state.id);
            if (writeResult) {
            // POST new friend
            //_friendsRepository.post()
            // Status.success
          } else {
            // Error
            emit(state.copyWith(
              status: AddFriendStatus.failure, 
              errorMessage: 'Failed to add your new friend'
            ));
            // State to retry
            emit(state.copyWith(
              status: AddFriendStatus.notFound,
              displayMessage: 'Friend not found. Try again ?'
            ));
          }
          }
        }
        print("Let role switching again");
        changingRole = true;
      });

      // Peripheral Stream Write Subscription
      final StreamSubscription peripheralSubscription = 
        _peripheralManager.writeRequestStream.listen((eventArgs) async {
          changingRole = false;
          final request = eventArgs.request;
          final value = utf8.decode(request.value);
          print('(Peripheral) Value received : $value');
          emit(state.copyWith(displayMessage: '(Peripheral) Value received : $value'));
          if (value == state.friendName) {
            // Emit status to break while loop
            emit(state.copyWith(status : AddFriendStatus.found));
            // If correct, respond to request, 
            await _peripheralManager.respondWrite(request);
            // POST new friend
            //_friendsRepository.post()
            // Status.success
          } else {
            await _peripheralManager.respondWriteWithError(request);
          }
          changingRole = true;
      });
      
      // Loop break on status change
      while (state.status == AddFriendStatus.detection) {
        // Launch central        
        if (changingRole && role && !_centralManager.discovering) {
          print('Central');
          emit(state.copyWith(displayMessage: "Searching (Central)"));
          await _peripheralManager.stopAdvertising();
          await _centralManager.startDiscovery();
          // Delay
          await Future.delayed(Duration(seconds: 5), () {
            role = false;
          });
        }
        // Launch Peripheral
        if(changingRole && !role && !_peripheralManager.advertising) {
          print('Peripheral');
          emit(state.copyWith(displayMessage: "Searching (Peripheral)"));

          await _centralManager.stopDiscovery();
          await _peripheralManager.startAdvertising(state.id);
          // Delay
          await Future.delayed(Duration(milliseconds: (5000 + Random().nextInt(5000))), () {
            role = true;
            maxTurn --;
          });
        }
        // Breaking loop after maxTurn 
        if (maxTurn == 0) {
          // State to retry
          emit(state.copyWith(
            status: AddFriendStatus.notFound,
            displayMessage: 'Friend not found. Try again ?'
          ));
        }
      }
      // State to retry
      emit(state.copyWith(
        status: AddFriendStatus.notFound,
        displayMessage: 'Friend not found. Try again ?'
      ));
      // Cancel subscriptions
      if (state.status == AddFriendStatus.found) {
        _centralManager.dispose();
        _peripheralManager.dispose();
      }
      centralSubscription.cancel();
      peripheralSubscription.cancel();
    }
    
And both my BleCentral and BlePeripheral are based on the ViewModel of the exemple. Sorry if my question is too trivial or doesn't have her place here, I'm still a bit new in the coding community.

Thank you for your answer and the librairy, which is helping me a lot.

yanshouwang, 2024-12-21:
Typically we don't act both central role and peripheral role at the same time.

I see there are some methods of the _centralManager does not belong to this plugin.

I don not suggest to change roles in this way. There should always be a peripheral act as GATT server. And other devices use the CentralManager act as GATT client to communicate with the GATT server

RobinSteiger, 2024-12-21:
Thank you for your response. 

What I'm trying to achieve is a connection and "hand shake" between 2 mobile without asking the user to choose between being the central or the peripheral, just a way to add another user close to him physically as a friend automatically. I don't need any more connection or functionality than a write between a machine and the other.
So I've thought that launching each role until one detect another would define the role of each one, with one always acting as a central and the other as a peripheral. 

Do you think there would be another way to achieve this connection without changing the roles ? Maybe be launching the two roles together and stopping the other role when one begin to work ? If the split is absolutely necessary, I'll do it but it will be way more unintuitive for the user I think.

Also, I've taken the method connect, discoverGatt(which include discovering GATT services and characteristics) and writeRequest from the PeripheralViewModel to put them into the CentralManager as they are the only method I need to achieve my handshake, could it be a source of problems ? Is it necessary to split the discovery of devices and the connection/write request ?

Thanks again, I admit it make quite some time that I'm stuck on this functionality and not progressing any more.

yanshouwang, 2024-12-23:
@RobinSteiger I have no idea about this serverless situation. But I think you can try to seprate the two roles instead of switch between them. Run central and peripheral roles in parallel on newest device should be ok.

RobinSteiger, 2024-12-23:
@yanshouwang It was this idea that I hadn't, it's working perfectly now. Thanks again for your help and library.

closed 2024-12-23 by RobinSteiger