yanshouwang/bluetooth_low_energy #117: Donation

Jeff-Lawson, 2024-12-28:
Hello Yanshou,

Are you able to accept BitCoin donations? If so, please let us know your BitCoin Address  :-)

Warmest regards,

Jeff

yanshouwang, 2024-12-29:
Hi, Jeff

Thanks for your kindness, here is my Bitcoin address.

[image]

I'll add the donation part to the README a few days later.

Jeff-Lawson, 2024-12-30:
> Links:

Hi Yanshou,

I tried to pay but my BitCoin service wanted:

    - Wallet host (e.g., self-hosted, ChangeHero)

    - Wallet address, i.e. not BitCoin address!

Please see attached screenshot.

Apologies. Are you able to provide that information?

Warmest regards,

Jeff

yanshouwang, 2025-01-04:
> Hi Yanshou, My BitCoin service uses wallets but I was able to use another service that accepted QR Codes (the Coinbase app for iOS). Donation now sent :-) Thank you very much for your hard and fruitful work on bluetooth_low_energy, it is very much appreciated :-) Warmest regards, Jeff

I'v received the donation, thanks again for your kindness! I'll still working hard on this project in the future. 

I'll add the sponsors part in the README to show the details of this donation later.

closed 2025-01-05 by yanshouwang

Jeff-Lawson, 2025-01-06:
Hi Yanshou,

I really appreciate the hard work you put into bluetooth_low_energy.

If you ever get around to completing PeripheralManager for Linux, I will 
definitely send the same donation again  :-)

Warmest regards,

Jeff

yanshouwang, 2025-01-06:
> Hi Yanshou, I really appreciate the hard work you put into bluetooth_low_energy. If you ever get around to completing PeripheralManager for Linux, I will definitely send the same donation again :-) Warmest regards, Jeff

Thanks again for your generous donation. You have already helped a lot. Support for Linux is part of the development plan. Currently, The Linux support is based on a wrapper for the [bluez](https://pub.dev/packages/bluez) plugin. However, it does not support the peripheral's API yet. Perhaps I could try to implement this based on the [dbus](https://pub.dev/packages/dbus) plugin.

I will investigate the possibility of the dbus impl in the next months.

Jeff-Lawson, 2025-01-07:
> Linux support is based on a wrapper for the bluez [1] plugin. However,
> implement this based on the dbus [2] plugin.

Hi Yanshou,

Perhaps you forgot; bluez does now support advertising:
    https://github.com/yanshouwang/bluetooth_low_energy/issues/79

Here's the documentation:
    https://bluez-peripheral.readthedocs.io/en/rework-docs/
    https://bluez-peripheral.readthedocs.io/en/stable/

Warmest regards,

Jeff