yanshouwang/bluetooth_low_energy #118: The BluetoothLowEnergyDarwinPlugin ` registerWith ` method whether can delay initialization?

hpzerozero, 2025-01-02:
[image]
The BluetoothLowEnergyDarwinPlugin ` registerWith ` method whether can delay initialization?Because in registration initialization, permissions are called once the library is added, but I only want to call permissions where I need them.

yanshouwang, 2025-01-02:
The Android permission is not requested automatically after V6.0, you need call authorize manually now.
The iOS permission is requested automatically by system when the app run the first time. It can not be requested manually on iOS

hpzerozero, 2025-01-02:
> The Android permission is not requested automatically after V6.0, you need call authorize manually now. The iOS permission is requested automatically by system when the app run the first time. It can not be requested manually on iOS

I can do this when developing native iOS, only request Bluetooth permissions when you need Bluetooth. Does the plugin have to request Bluetooth permission as soon as it launches the application? Is that the only way to design it?

yanshouwang, 2025-01-02:
> I can do this when developing native iOS, only request Bluetooth permissions when you need Bluetooth. Does the plugin have to request Bluetooth permission as soon as it launches the application? Is that the only way to design it?

The BluetoothManager is a singleton now and created when the app started. I think I can redesign this singleton pattern to request the permission as needed.

hpzerozero, 2025-01-02:
> The Android permission is not requested automatically after V6.0, you need call authorize manually now. The iOS permission is requested automatically by system when the app run the first time. It can not be requested manually on iOS

Since iOS automatically requests Bluetooth permission  when App initializes the CentralManager or PeripheralManager object . Hence, I think if you don't put `MyCentralManager()` in the `void registerWith` method and provide an additional manual registration method, you maybe can solve this problem. Since Bluetooth is only a small part of our app, I don't want users to request Bluetooth permission as soon as they open it.

hpzerozero, 2025-01-02:
> The BluetoothManager is a singleton now and created when the app started. I think I can redesign this singleton pattern to request the permission as needed.

Thank you for your reply and look forward to the new version!

hpzerozero, 2025-01-02:
> I understand what you need. I will make the native instance created when you create the manager on the Dart side in the next version.

Thank you very much！

closed 2025-07-29 by yanshouwang