yanshouwang/bluetooth_low_energy #124: iOS 平台如果service是 平台级别的，应用无法获取去特征值的属性值

windstylebar, 2025-02-18:
`PlatformException(Error Domain=CBErrorDomain Code=8 "The specified UUID is not allowed for this operation." UserInfo={NSLocalizedDescription=The specified UUID is not allowed for this operation.}, NSError, 
Stacktrace: [
"0   bluetooth_low_energy_darwin         0x00000001061d13a4 $s27bluetooth_low_energy_darwin9wrapError33_D3FF39C2E9589F2A07964E0F3A1AEE9ELLySayypSgGypF + 604", 
"1   bluetooth_low_energy_darwin         0x00000001061d4598 $s27bluetooth_low_energy_darwin28MyCentralManagerHostAPISetupC5setUp15binaryMessenger3api20messageChannelSuffixySo013FlutterBinaryM0_p_AA0efgH3API_pSgSStFZyypSg_yAKctcfU9_ys6ResultOySayAA0E15GATTServiceArgsVGs5Error_pGcfU_Tm + 236", 
"2   bluetooth_low_energy_darwin         0x00000001061e0fa0 $s27bluetooth_low_energy_darwin28MyCentralManagerHostAPISetupC5setUp15binaryMessenger3api20messageChannelSuffixySo013FlutterBinaryM0_p_AA0efgH3API_pSgSStFZyypSg_yAKctcfU12_ys6ResultOySayAA0E18GATTDescriptorArgsVGs5Error_pGcfU_TA + 28", 
"3   bluetooth_low_energy_darwin         0x00000001061e1018 $ss6ResultOySay27bluetooth_low_energy_darwin24MyGATTCharacteristicArgsVGs5Error_pGIegg_AHytIegnr_TRTA + 32", 
"4   bluetooth_low_energy_darwin         0x00000001061e4374 $s27bluetooth_low_energy_darwin16MyCentralManagerC22didDiscoverDescriptors10peripheral14characteristic5errorySo12CBPeripheralC_So16CBCharacteristicCs5Error_pSgtF + 344", 
"5   bluetooth_low_energy_darwin         0x00000001061e8eec $s27bluetooth_low_energy_darwin20MyPeripheralDelegateC10peripheral_11didReadRSSI5errorySo12CBPeripheralC_So8NSNumberCs5Error_pSgtFToTm + 116", 
"6   CoreBluetooth                       0x00000001bdbe16dc AB215002-518E-3446-8FE5-45397877F1C0 + 313052", 
"7   CoreBluetooth                       0x00000001bdbe180c AB215002-518E-3446-8FE5-45397877F1C0 + 313356", 
"8   CoreBluetooth                       0x00000001bdbde35c AB215002-518E-3446-8FE5-45397877F1C0 + 299868", 
"9   CoreBluetooth                       0x00000001bdb9b788 AB215002-518E-3446-8FE5-45397877F1C0 + 26504", 
"10  CoreBluetooth                       0x00000001bdb9b6b4 AB215002-518E-3446-8FE5-45397877F1C0 + 26292", 
"11  CoreBluetooth                       0x00000001bdb9b560 AB215002-518E-3446-8FE5-45397877F1C0 + 25952", 
"12  libdispatch.dylib                   0x00000001072c8a30 _dispatch_call_block_and_release + 32", 
"13  libdispatch.dylib                   0x00000001072ca71c _dispatch_client_callout + 20", 
"14  libdispatch.dylib                   0x00000001072d25e8 _dispatch_lane_serial_drain + 828", 
"15  libdispatch.dylib                   0x00000001072d3394 _dispatch_lane_invoke + 460", 
"16  libdispatch.dylib                   0x00000001072dacbc _dispatch_main_queue_drain + 776", 
"17  libdispatch.dylib                   0x00000001072da9a4 _dispatch_main_queue_callback_4CF + 44", 
"18  CoreFoundation                      0x00000001956dc204 1532D3D8-9B3B-3F2F-B35F-55A20DDF411B + 352772", 
"19  CoreFoundation                      0x00000001956d9440 1532D3D8-9B3B-3F2F-B35F-55A20DDF411B + 341056", 
"20  CoreFoundation                      0x00000001956d8830 CFRunLoopRunSpecific + 588", 
"21  GraphicsServices                    0x00000001e16c81c4 GSEventRunModal + 164", 
"22  UIKitCore                           0x000000019824c07c 9DD41986-B4C2-3A25-9995-33FED2DD9A89 + 4014204", 
"23  UIKitCore                           0x00000001982fa8a0 UIApplicationMain + 340", 
"24  mysigen.debug.dylib                 0x00000001080d7c30 __debug_main_executable_dylib_entry_point + 64", 
"25  dyld                                0x00000001bb0ceec8 0ACAB10C-315E-3225-BD57-9290D6CC3734 + 212680"], null)
      #0      MyCentralManagerHostAPI.discoverDescriptors (package:bluetooth_low_energy_darwin/src/my_api.g.dart:807:7)`

对于系统级别的api调用discoverCharacteristics 时，会有上述异常，可以在`bluetooth_low_energy_darwin/lib/src/my_central_manager.dart`
能否在做一下修改
````dart
logger.info('discoverDescriptors: $uuidArgs.$hashCodeArgs');
    try {
    final descriptorsArgs = await _api
        .discoverDescriptors(uuidArgs, hashCodeArgs)
        .then((args) => args.cast<MyGATTDescriptorArgs>());
    return descriptorsArgs;
    } catch (e) {
      return [];
    }
  }
````

yanshouwang, 2025-02-18:
下个版本会添加 discoverServices, discoverCharacteristics, discoverDescriptors 这三个方法，iOS 平台可以根据需要只发现需要的目标，是否能满足此需求?

ArdWang, 2025-03-28:
final services = await _manager.discoverGATT(model.peripheral!);    我用这个为什么读取不出来 services Macos系统 13.0以上