yanshouwang/bluetooth_low_energy #134: Windows - Advertising data very low frequency

tototo23, 2025-08-22:
Hi,
First, congrats for this lib !
I am trying to find a BLE lib to migrate to, from an other library i was using, because on Windows, the connexion step was not stable, neither the reception of advertising data frequency on Windows. Yours looks more stable and promising !
- the connexion is very stable with yours
- but for now, the advertising scan is still "slow" on Windows

On Android/iOS, the reception of advertising data is very quick as expected (my BLE products send advertising data multiple times/seconds), but on Windows, the scan receives BLE advertising frames sometimes every 30 seconds for a product... so 97% of frames are "lost".

Is there a way to add an option to make the scan mode more "agressive" on Windows ? I've read that on windows, by default, it's an optimization for energy consumption that the scan is slow on advertising data. But if we need explcitely to read more real-time advertising data, it would be intresting to have this option.

Thanks again

yanshouwang, 2025-08-25:
The scanning is not slow in my test. You can use the example to mock a peripheral on a device, and run the example on another windows device to scan that device, in my test, it's pretty fast.

The only thing I can find is the [ScanningMode](https://learn.microsoft.com/zh-cn/uwp/api/windows.devices.bluetooth.advertisement.bluetoothlescanningmode?view=winrt-26100) property in the BluetoothLEAdvertisementWatcher class. It's has been set to Active in this plugin.

I'm wondering if your BLE products have sent the scan response? if you don't, maybe something is ignored? See https://github.com/yanshouwang/bluetooth_low_energy/blob/79fc7af501453ca1b4114ac9f6e14fba5e12ed4a/bluetooth_low_energy_windows/lib/src/my_central_manager.dart#L297

You can debug this line, or set `Logger.root.level = Level.ALL;` to see if your advertisement is ignored.

tototo23, 2025-08-25:
@yanshouwang I've got some updates :)

I've tried adding some debug informations, and discovered that if often goes in the `if (ignored)` condition at the line you mentionned. I've checked also, it's because the type of those frames are `MyAdvertisementTypeArgs.connectableUndirected` and as you said, it's waiting a scanResponse.

In my debug logs, i've seen that the product sometimes answers with a `scanResponse` after the `connectableUndirected`. but it looks very random... I think it's the reason why, when no scanResponse, i don't have the advertising frame at all. Would it be possible to have a scan option, disabled by default, to not wait for a ScanResponse ?

yanshouwang, 2025-08-26:
> [@yanshouwang](https://github.com/yanshouwang) I've got some updates :)
>
> I've tried adding some debug informations, and discovered that if often goes in the `if (ignored)` condition at the line you [...]

The windows is weired when handling the advertisement. On other platforms the advertisements with different type with the same device is combined by the OS and we don't need to do this waiting logic. But on windows the advertisements is received one by one, I have to try to combine the advertisements by myself, you can see there is a `TODO` comment on the ignore logic. Maybe I should not try to combine the advertisement by myself as it seems to be a complicated thing, but this will break the advertisement API and make it different with other platforms.

Or can you make the scan response more reliable? You can test with the example to see the difference with your products.